xdspacelab/openvslam #11: Docker build not compiling.

amitahire, 2019-05-22:
Hello,

I was trying to build from Docker but I couldnt get it to build on MacOSX

For g2o it couldnt find Qt5 but I guess that shouldnt stop the compilation... It was stuck for a long time at 41% mark.

Any pointers would be appreciated, Thanks. 

```
➜  openvslam git:(master)               docker build -t openvslam-desktop .
Sending build context to Docker daemon  6.377MB
Step 1/36 : FROM ubuntu:18.04
 ---> 7698f282e524
Step 2/36 : RUN set -x &&   apt update -y -qq &&   apt upgrade -y -qq --no-install-recommends &&   : "basic dependencies" &&   apt install -y -qq     build-essential     pkg-config     cmake     git     wget     curl     tar     unzip &&   : "g2o dependencies" &&   apt install -y -qq     libgoogle-glog-dev     libatlas-base-dev     libsuitesparse-dev &&   : "OpenCV dependencies" &&   apt install -y -qq     libgtk-3-dev     libjpeg-dev     libpng++-dev     libtiff-dev     libopenexr-dev     libwebp-dev     ffmpeg     libavcodec-dev     libavformat-dev     libavutil-dev     libswscale-dev     libavresample-dev &&   : "Pangolin dependencies" &&   apt install -y -qq     libglew-dev &&   : "other dependencies" &&   apt install -y -qq     libyaml-cpp-dev &&   : "remove cache" &&   apt autoremove -y -qq &&   rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 4fb49f7fed8a
Step 3/36 : ARG CMAKE_INSTALL_PREFIX=/usr/local
 ---> Using cache
 ---> 1b28ad7e8108
Step 4/36 : ARG NUM_THREADS=4
 ---> Using cache
 ---> b49e3a994442
Step 5/36 : ENV CPATH=${CMAKE_INSTALL_PREFIX}/include:${CPATH}
 ---> Using cache
 ---> d275c719466f
Step 6/36 : ENV C_INCLUDE_PATH=${CMAKE_INSTALL_PREFIX}/include:${C_INCLUDE_PATH}
 ---> Using cache
 ---> b902a01664d7
Step 7/36 : ENV CPLUS_INCLUDE_PATH=${CMAKE_INSTALL_PREFIX}/include:${CPLUS_INCLUDE_PATH}
 ---> Using cache
 ---> b9257d47c8d2
Step 8/36 : ENV LIBRARY_PATH=${CMAKE_INSTALL_PREFIX}/lib:${LIBRARY_PATH}
 ---> Using cache
 ---> 6df63ec205b7
Step 9/36 : ENV LD_LIBRARY_PATH=${CMAKE_INSTALL_PREFIX}/lib:${LD_LIBRARY_PATH}
 ---> Using cache
 ---> 6963e8d905b9
Step 10/36 : ENV NVIDIA_VISIBLE_DEVICES ${NVIDIA_VISIBLE_DEVICES:-all}
 ---> Using cache
 ---> bb95a2280a0a
Step 11/36 : ENV NVIDIA_DRIVER_CAPABILITIES ${NVIDIA_DRIVER_CAPABILITIES:+$NVIDIA_DRIVER_CAPABILITIES,}graphics
 ---> Using cache
 ---> 56569bcc425c
Step 12/36 : ARG EIGEN3_VERSION=3.3.7
 ---> Using cache
 ---> 41fac59254f9
Step 13/36 : WORKDIR /tmp
 ---> Using cache
 ---> d6775ccfe074
Step 14/36 : RUN set -x &&   wget -q http://bitbucket.org/eigen/eigen/get/${EIGEN3_VERSION}.tar.bz2 &&   tar xf ${EIGEN3_VERSION}.tar.bz2 &&   rm -rf ${EIGEN3_VERSION}.tar.bz2 &&   mv eigen-eigen-* eigen-${EIGEN3_VERSION} &&   cd eigen-${EIGEN3_VERSION} &&   mkdir -p build &&   cd build &&   cmake     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}     .. &&   make -j${NUM_THREADS} &&   make install &&   cd /tmp &&   rm -rf *
 ---> Using cache
 ---> e5e4b5f96be7
Step 15/36 : ENV Eigen3_DIR=${CMAKE_INSTALL_PREFIX}/share/eigen3/cmake
 ---> Using cache
 ---> dd6e6959aa5e
Step 16/36 : ARG G2O_COMMIT=9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a
 ---> Using cache
 ---> 1e2a8c40b6e0
Step 17/36 : WORKDIR /tmp
 ---> Using cache
 ---> cb4f781486ed
Step 18/36 : RUN set -x &&   git clone https://github.com/RainerKuemmerle/g2o.git &&   cd g2o &&   git checkout ${G2O_COMMIT} &&   mkdir -p build &&   cd build &&   cmake     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}     -DBUILD_SHARED_LIBS=ON     -DBUILD_UNITTESTS=OFF     -DBUILD_WITH_MARCH_NATIVE=ON     -DG2O_USE_CHOLMOD=ON     -DG2O_USE_CSPARSE=ON     -DG2O_USE_OPENGL=OFF     -DG2O_USE_OPENMP=ON     -DG2O_BUILD_APPS=OFF     -DG2O_BUILD_EXAMPLES=OFF     -DG2O_BUILD_LINKED_APPS=OFF     .. &&   make -j${NUM_THREADS} &&   make install &&   cd /tmp &&   rm -rf *
 ---> Running in 430e0cfa5113
+ git clone https://github.com/RainerKuemmerle/g2o.git
Cloning into 'g2o'...
+ cd g2o
+ git checkout 9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a
Note: checking out '9b41a4ea5ade8e1250b9c1b279f3a9c098811b5a'.

You are in 'detached HEAD' state. You can look around, make experimental
[... 147 lines cut ...]
[ 48%] Building CXX object g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2_twopointsxy.cpp.o
[ 48%] Building CXX object g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_se2_lotsofxy.cpp.o
[ 49%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/parameter_camera.cpp.o
[ 50%] Linking CXX shared library ../../../../lib/libg2o_solver_pcg.so
[ 50%] Built target solver_pcg
[ 51%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/parameter_stereo_camera.cpp.o
[ 52%] Building CXX object g2o/types/slam2d/CMakeFiles/types_slam2d.dir/edge_xy_prior.cpp.o
[ 53%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz.cpp.o
[ 53%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz_disparity.cpp.o
[ 54%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_pointxyz_depth.cpp.o
[ 55%] Linking CXX shared library ../../../../lib/libg2o_types_slam2d.so
[ 55%] Built target types_slam2d
[ 56%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_prior.cpp.o
[ 57%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_xyzprior.cpp.o
[ 57%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_se3_lotsofxyz.cpp.o
[ 58%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_xyz_prior.cpp.o
[ 59%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/edge_pointxyz.cpp.o
[ 60%] Building CXX object g2o/types/slam3d/CMakeFiles/types_slam3d.dir/types_slam3d.cpp.o
[ 60%] Linking CXX shared library ../../../../lib/libg2o_types_slam3d.so
[ 60%] Built target types_slam3d
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
The command '/bin/sh -c set -x &&   git clone https://github.com/RainerKuemmerle/g2o.git &&   cd g2o &&   git checkout ${G2O_COMMIT} &&   mkdir -p build &&   cd build &&   cmake     -DCMAKE_BUILD_TYPE=Release     -DCMAKE_INSTALL_PREFIX=${CMAKE_INSTALL_PREFIX}     -DBUILD_SHARED_LIBS=ON     -DBUILD_UNITTESTS=OFF     -DBUILD_WITH_MARCH_NATIVE=ON     -DG2O_USE_CHOLMOD=ON     -DG2O_USE_CSPARSE=ON     -DG2O_USE_OPENGL=OFF     -DG2O_USE_OPENMP=ON     -DG2O_BUILD_APPS=OFF     -DG2O_BUILD_EXAMPLES=OFF     -DG2O_BUILD_LINKED_APPS=OFF     .. &&   make -j${NUM_THREADS} &&   make install &&   cd /tmp &&   rm -rf *' returned a non-zero code: 2
➜  openvslam git:(master)
```

shinsumicco, 2019-05-23:
Thank you for your interest.
I think it seems to memory shortage of Docker on mac.
Could you change `NUM_THREADS` to `1` in `Dockerfile` and incease the memory of the docker-machine, then rebuild the image?

shinsumicco, 2019-05-23:
https://github.com/xdspacelab/openvslam/blob/master/Dockerfile#L47
Please change the number of parallel builds from 4 to 1.

closed 2019-06-25 by shinsumicco