xdspacelab/openvslam #12: Why is there no reference to ORB_SLAM2 ?

nikolaseu, 2019-05-22:
I'm currently trying to learn about SLAM on my free time, so I'm looking at different open source projects. I noticed the core of this project is *really* close to ORB_SLAM2 but with different namespace and some change in the name of methods. Are the projects related in some way? Are both based on some other previous code base?

kensakurada, 2019-05-23:
We appreciate your letting us know about that we forgot to describe the references in the documentation (the details and the references are described in our paper that is under review). We compared and analyzed many kinds of open sources, such as ORB-SLAM, ProSLAM, and UcoSLAM. From the insights, we created our open source _from scratch_. Although the names of some variables and functions in OpenVSLAM might be close to them because both of OpenVSLAM and them are based on a feature-based algorithm, the code structures of other main parts (e.g. camera model, initializer, tracker, optimizer) are completely different. Considering your thoughtful comment, we added the references into [the documentation](https://openvslam.readthedocs.io/en/master/overview.html). Thank you for your cooperation.

closed 2019-05-23 by kensakurada

masahi, 2019-05-28:
I would suggest that you should at least add a proper acknowledgement or note to ORBSLAM2. Having spent considerable time on the ORBSLAM2 codebase, I'm afraid it is obvious that this codebase is directy derived from ORBSLAM2 (although written from scratch). I've read initialization, local_mapping and matching components, and I can see line by line correspondence between the two codebases.

kensakurada, 2019-05-28:
Thank you for your thoughtful comments. We added reference to README so that users notice that OpenVSLAM refers ORB-SLAM, ProSLAM and other SLAM systems at a glance. Please refer to the preprint of OpenVSLAM paper for the detail that will be uploaded soon.