xdspacelab/openvslam #8: Remove unnecessary null pointer checks

elfring, 2019-05-22:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
* [config](https://github.com/xdspacelab/openvslam/blob/19fecdaee07fc87f8bda884af21ffa88986ad4bc/src/openvslam/config.cc#L45 "Constructor for config")
* [initializer::reset](https://github.com/xdspacelab/openvslam/blob/19fecdaee07fc87f8bda884af21ffa88986ad4bc/src/openvslam/initialize/initializer.cc#L25 "initializer::reset function")

shinsumicco, 2019-05-22:
Thanks! I'll fix it.

closed 2019-05-23 by shinsumicco

elfring, 2019-05-24:
Thanks for [your small source code improvement](https://github.com/xdspacelab/openvslam/commit/70f34f6e16f7ef66949b8fbbbc97fc05e50007b5 "Removal of four unnecessary checks").

PR #10 (2019-05-22): Remove unnecessary null checks of pointers before deleting them
close #8