xdspacelab/openvslam #9: Improve exception safety with smart pointers

elfring, 2019-05-22:
Would you like to [wrap any pointer data members](https://github.com/xdspacelab/openvslam/blob/19fecdaee07fc87f8bda884af21ffa88986ad4bc/src/openvslam/track/tracker.cc#L22 "Update candidate: tracker class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

shinsumicco, 2019-05-22:
I'm planning to use `std::shared_ptr` and `std::weak_ptr` for some database objects.
(`std::unique_ptr` cannot be used for most of the objects because their resources are shared by multiple pointers.)
However, I intend to use raw pointers for landmark objects and keyfreme objects because `std::shared_ptr` contains the overhead due to reference counting.
Do you have any thoughts?

elfring, 2019-05-22:
Do any of the mentioned classes work without reference counting for specific objects already?

shinsumicco, 2019-05-26:
Could you tell me about it in more detail?
You mean some objects can be accessed via the reference variables, don't you?

elfring, 2019-05-26:
My knowledge about this software implementation is incomplete so that I am unsure which pointer variables should refer to shared objects finally.
[Is a class like “`tracker`” still an interesting update candidate](https://github.com/xdspacelab/openvslam/blob/19fecdaee07fc87f8bda884af21ffa88986ad4bc/src/openvslam/track/tracker.h#L150 "Update candidate")?

closed 2019-09-11 by shinsumicco