xdspacelab/openvslam #18: Does this support 360-degree stitching after image input?

hunkyu, 2019-05-24:
Hello, I tried it. I don't know if it is in the requirectangular piece. Is it the result of supporting multiple images after stitching together as a system input? Of course, the internal parameters after stitching have been calibrated.

shinsumicco, 2019-05-24:
Thank you for your interest.
Our system currently supports equirectangular images even which are combined of multiple images.
In fact, the sample equirectangular video files we provide for the tutorial are created from **dual fisheye** movies like [this](https://developers.theta360.com/intl/common/img/dualfisheye.bmp).
Does that make sense?

hunkyu, 2019-05-24:
[image]
Camera.name: "RICOH THETA S 960"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fps: 10.0
Camera.cols: 7168
Camera.rows: 930

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.84, 1.0]
  - [0.0, 0.2, 0.7, 1.0]
  - [0.8, 1.0, 0.7, 1.0]
I can't complete initialization with my own panoramic image.What do you think is the cause?

hunkyu, 2019-05-24:
I changed Camera.fps to 30, which can be initialized normally, but I found that I did not complete the closed loop, and the estimation in the z direction is not accurate.

shinsumicco, 2019-05-25:
Could you add margins on the top and on the bottom of images to make them be **2:1**?
Like this:

[image]

Accordingly adjust the `mask_rectangles` of the config file:

```
Feature.mask_rectangles:
- [0.0, 1.0, 0.0, 0.37]
- [0.0, 1.0, 0.63, 1.0]
```

If tracking failed soon after initialization, please downsize images to ~2K size.
And please confirm that the projection model of the camera you used for capturing is equirectangular.

hunkyu, 2019-05-25:
[image]
After the change, it is really better! But the gesture displayed in pangolin is very strange, the posture of the camera is no longer right.

hunkyu, 2019-05-25:
Feel that the heading angle has a fixed deviation

shinsumicco, 2019-05-25:
I'm glad to hear that!

The camera poses displayed in the pangolin viewer seem to be correct. It is because the front direction of your camera is diagonal to the moving direction.
The front direction of the equirectangular camera model we implemented is defined as the center of an image. The center of your input image is not the moving direction, isn't it?

hunkyu, 2019-05-25:
[image]
The third part of my image is the direction of movement of the camera, which is right in front.

hunkyu, 2019-05-25:
[image]
I tested the effect of relocation, the effect is very good, I don't know if it works normally under different lighting changes! Also, I found out why your relocation program is so costly, up to 400%. Can you analyze the specific reasons? I feel that the positioning program multi-threading is to track the association with dbow parsing, and does not need to create local maps and closed-loop detection.

hunkyu, 2019-05-26:
What is the basis for the Feature.mask_rectangles setting? @shinsumicco

closed 2019-09-11 by shinsumicco