xdspacelab/openvslam #19: compile error

Cavendish-Koo, 2019-05-24:
when I compiled the code, I got errors like these
/home/perm1ss4on/github/openvslam/src/openvslam/optimize/global_bundle_adjuster.cc:35:26: error: ‘make_unique’ is not a member of ‘g2o’
     auto linear_solver = ::g2o::make_unique<::g2o::LinearSolverCSparse<::g2o::BlockSolver_6_3::PoseMatrixType>>();
                          ^
/home/perm1ss4on/github/openvslam/src/openvslam/optimize/global_bundle_adjuster.cc:35:110: error: expected primary-expression before ‘>’ token
     auto linear_solver = ::g2o::make_unique<::g2o::LinearSolverCSparse<::g2o::BlockSolver_6_3::PoseMatrixType>>();
                                                                                                              ^
/home/perm1ss4on/github/openvslam/src/openvslam/optimize/global_bundle_adjuster.cc:35:113: error: expected primary-expression before ‘)’ token
     auto linear_solver = ::g2o::make_unique<::g2o::LinearSolverCSparse<::g2o::BlockSolver_6_3::PoseMatrixType>>();

I follow the tutorial and I download the newest g2o, is there anyting I have missed?

shinsumicco, 2019-05-25:
Could you tell me your environment? (OS, compiler, etc)

Cavendish-Koo, 2019-05-25:
yes.I used gcc 5.4 compile the code on ubuntu 16.04.I have rebuild g2o and it doesn't work.

shinsumicco, 2019-05-25:
Umm... I'm not sure about the reason of the compile error.
Please insert `#include <g2o/stuff/misc.h>` to [`global_bundle_adjuster.cc`](https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/optimize/global_bundle_adjuster.cc#L9), and post the rebuild log here?

Cavendish-Koo, 2019-05-25:
After insert the code, I rebuild it.

the cmake log:
-- Build type: Release
-- Address/Memory sanitizer: DISABLED
-- Stack trace logger: ENABLED
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Found Glog: /usr/include  
-- Google Perftools: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- C++11 support: OK (-std=c++11)
-- Use OpenCV 3.4.4
-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- A library with BLAS API found.
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found CHOLMOD library: /usr/lib/x86_64-linux-gnu/libcholmod.so
-- Found CHOLMOD header in: /usr/include/suitesparse
-- Found SuiteSparseQR library: /usr/lib/x86_64-linux-gnu/libspqr.so
-- Found SuiteSparseQR header in: /usr/include/suitesparse
-- Found Intel Thread Building Blocks (TBB) library: /usr/lib/x86_64-linux-gnu/libtbb.so, assuming SuiteSparseQR was compiled with TBB.
-- Found Intel Thread Building Blocks (TBB) Malloc library: /usr/lib/x86_64-linux-gnu/libtbbmalloc.so
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found yaml-cpp library: /usr/lib/x86_64-linux-gnu/libyaml-cpp.so
-- Found yaml-cpp header in: /home/perm1ss4on/anaconda3/include
-- BoW framework: DBoW2 (found in /usr/local/include/DBoW2)
-- Viewer for examples: pangolin viewer
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/perm1ss4on/github/openvslam/build/googletest-download
Scanning dependencies of target googletest-download
[ 11%] Creating directories for 'googletest-download'
[ 22%] Performing download step (git clone) for 'googletest-download'
Cloning into 'googletest-src'...
Already on 'master'
Your branch is up-to-date with 'origin/master'.
[ 33%] No patch step for 'googletest-download'
[ 44%] Skipping update step for 'googletest-download'
[ 55%] No configure step for 'googletest-download'
[ 66%] No build step for 'googletest-download'
[ 77%] No install step for 'googletest-download'
[ 88%] No test step for 'googletest-download'
[100%] Completed 'googletest-download'
[100%] Built target googletest-download
-- Found PythonInterp: /home/perm1ss4on/anaconda3/bin/python (found version "3.6.5") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/perm1ss4on/github/openvslam/build

It seems that some dependencies are not found such as METIS,and CXSparse.I reinstall all the dependencies and all of them are newest. I also find libcxsparse.so.3.1.4 under  /usr/lib/x86_64-linux-gnu.

the build log:

Scanning dependencies of target test_helper
Scanning dependencies of target gtest
Scanning dependencies of target openvslam
[  1%] Building CXX object googletest-build/googletest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[  2%] Building CXX object test/helper/CMakeFiles/test_helper.dir/bearing_vector.cc.o
[... 352 lines cut ...]
                                              ^
In file included from /home/perm1ss4on/github/openvslam/src/openvslam/optimize/g2o/se3/perspective_pose_opt_edge.h:6:0,
                 from /home/perm1ss4on/github/openvslam/src/openvslam/optimize/g2o/se3/pose_opt_edge_wrapper.h:7,
                 from /home/perm1ss4on/github/openvslam/src/openvslam/optimize/pose_optimizer.cc:4:
/home/perm1ss4on/github/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:14:7: note:   because the following virtual functions are pure within ‘openvslam::optimize::g2o::se3::shot_vertex’:
 class shot_vertex final : public ::g2o::BaseVertex<6, ::g2o::SE3Quat> {
       ^
In file included from /usr/local/include/g2o/core/base_vertex.h:30:0,
                 from /usr/local/include/g2o/types/sba/types_six_dof_expmap.h:30,
                 from /home/perm1ss4on/github/openvslam/src/openvslam/util/converter.h:8,
                 from /home/perm1ss4on/github/openvslam/src/openvslam/data/frame.h:6,
                 from /home/perm1ss4on/github/openvslam/src/openvslam/optimize/pose_optimizer.cc:1:
/usr/local/include/g2o/core/optimizable_graph.h:330:22: note: 	virtual void g2o::OptimizableGraph::Vertex::oplusImpl(const double*)
         virtual void oplusImpl(const double* v) = 0;
                      ^
src/openvslam/CMakeFiles/openvslam.dir/build.make:854: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/optimize/graph_optimizer.cc.o' failed
[... cut ...]
make[2]: *** [src/openvslam/CMakeFiles/openvslam.dir/optimize/transform_optimizer.cc.o] Error 1
src/openvslam/CMakeFiles/openvslam.dir/build.make:902: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/optimize/pose_optimizer.cc.o' failed
make[2]: *** [src/openvslam/CMakeFiles/openvslam.dir/optimize/pose_optimizer.cc.o] Error 1
CMakeFiles/Makefile2:106: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/all' failed
make[1]: *** [src/openvslam/CMakeFiles/openvslam.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

shinsumicco, 2019-05-25:
It seems that the headers of `g2o` not correctly included.
Would you confirm that the headers of `g2o` is placed at the correct path?
Then, please specify the include path by `-DG2O_INCLUDE_DIR=/path/to/correct/include` for cmake when building OpenVSLAM.

Cavendish-Koo, 2019-05-25:
I modify the cmakelists:

if(G2O_FOUND)
    include_directories(${G2O_INCLUDE_DIR})
    message(STATUS "G2O FOUND UNDER ${G2O_INCLUDE_DIR}")

Then I got the output:
-- G2O FOUND UNDER /usr/local/include
which is the correct path.And when I build the code I get the same error.

I guess if CXSparse cause the error,because it cannot be found by findpackage and I link it myself:
set(CXSPARSE_INCLUDE_DIRS /usr/include/suitesparse/cxsparse)
set(CXSPARSE_LIBRARIES /usr/lib/x86_64-linux-gnu/libcxsparse.so.3.1.4)
include_directories(${CXSPARSE_INCLUDE_DIRS})

I follow the installation to install dependency: 
apt install -y libatlas-base-dev libsuitesparse-dev.

suuman, 2019-05-27:
I too had similar errors. However, I managed to solve by building g2o from the master branch. 
https://github.com/RainerKuemmerle/g2o

Since I haven't install g2o on /usr/local, I have done following to avoid conflict with the version that is installed in /usr/local

a) In CMakeLists.txt, I have manually specified G2O_ROOT
   set(G2O_ROOT my_g2o_install_path)

b) In cmake/FindG2O.cmake 
   I have added a line so that correct g2o include dir is used
  NO_DEFAULT_PATH
after line 39.. 
Now it looks like
FIND_PATH(G2O_INCLUDE_DIR NAMES g2o/core/base_vertex.h
        PATHS
        ${G2O_ROOT}/include
        $ENV{G2O_ROOT}/include
        $ENV{G2O_ROOT}
        /usr/local/include
        /usr/include
        /opt/local/include
        /sw/local/include
        /sw/include
        **NO_DEFAULT_PATH**
        )

Cavendish-Koo, 2019-05-27:
Thanks for your help. 
I used to install it by download zip.When I git clone to install it, no errors occur.Thank you very much!

closed 2019-05-27 by Cavendish-Koo