xdspacelab/openvslam #21: core dump

southpaw-wu, 2019-05-25:
When I compiled it and entered the command according to simple tutorial,The frame viewer and map viewer launched for 1s and exited, showing core dump段错误 (核心已转储).
Has anybody ever met?

southpaw-wu, 2019-05-25:
I've found a solution，configure and rebuild g2o and OpenVSLAM with -DBUILD_WITH_MARCH_NATIVE=OFF option for cmake.

shinsumicco, 2019-05-25:
Thanks!
The solution for the trouble you faced has been presented in [Trouble Shooting](https://openvslam.readthedocs.io/en/master/trouble_shooting.html) in the documentation.

closed 2019-05-25 by shinsumicco

joegochal, 2019-05-27:
I get a similar error about 10 seconds into the tutorial.  However, I'm running it from within the Docker image.  Is there a way to rebuild that image or offer an alternative that addresses this issue?  Or should I just build everything manually?  Thanks.