xdspacelab/openvslam #28: Problem in Linking on windows

RamadanAhmed, 2019-05-26:
Hi,
I build the project on windows 10 using visual studio 19, but there is some error in linking so if someone could help, it will be great.
error msg 
`LNK2005	"public: void __cdecl g2o::BaseVertex<6,class g2o::SE3Quat>::setEstimate(class g2o::SE3Quat const &)" (?setEstimate@?$BaseVertex@$05VSE3Quat@g2o@@@g2o@@QEAAXAEBVSE3Quat@2@@Z) already defined in shot_vertex.cc.obj`

`LNK2005	"public: virtual void __cdecl g2o::BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1> >::setMeasurement(class Eigen::Matrix<double,2,1,0,2,1> const &)" (?setMeasurement@?$BaseEdge@$01V?$Matrix@N$01$00$0A@$01$00@Eigen@@@g2o@@UEAAXAEBV?$Matrix@N$01$00$0A@$01$00@Eigen@@@Z) already defined in equirectangular_pose_opt_edge.cc.obj`

guttyon, 2019-05-29:
Hi. I've successed to build using visual studio 2019!
run_video_slam.exe and run_video_slam.exe are working on windows 10.
I've exprienced your error.
constructor code in equirectangular_pose_opt_edge.cc might be moved to class decralation in equirectangular_pose_opt_edge.h.
equirectangular_pose_opt_edge::equirectangular_pose_opt_edge() : ::g2o::BaseUnaryEdge<2, Vec2_t, shot_vertex>() {}

The same applies to shot_vertex.

RamadanAhmed, 2019-05-29:
I will try it and give feedback
Thank you

closed 2019-08-21 by shinsumicco

antithing, 2019-09-03:
Hi @RamadanAhmed @guttyon , I am having this same issue on windows, (as well as a few linker errors). Would you be able to please describe the changes that you made exactly?

Thank you!