xdspacelab/openvslam #34: OpenVslam build 'make' error in ubuntu16.04

ZQxiaoshi, 2019-05-28:
Thank you for your great work! 
I have installed the enviroment as required, but when I run the 'make' command in the build folder of openvslam, the terminal window display error. Details are as follows：
...........
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/type.h:10,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:4,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.cc:1:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
In file included from /home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.cc:1:0:
/home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:28:26: error: ‘number_t’ does not name a type
     void oplusImpl(const number_t* update_) override {
                          ^
/home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:28:10: error: ‘void openvslam::optimize::g2o::se3::shot_vertex::oplusImpl(const int*)’ marked ‘override’, but does not override
     void oplusImpl(const number_t* update_) override {
          ^
/home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h: In member function ‘void openvslam::optimize::g2o::se3::shot_vertex::oplusImpl(const int*)’:
/home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:29:48: error: no matching function for call to ‘Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >::Map(const int*&)’
         Eigen::Map<const Vec6_t> update(update_);
                                                ^
In file included from /usr/local/include/eigen3/Eigen/Core:449:0,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/type.h:10,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.h:4,
                 from /home/zq/catkin_ws/src/openvslam/src/openvslam/optimize/g2o/se3/shot_vertex.cc:1:
/usr/local/include/eigen3/Eigen/src/Core/Map.h:149:12: note: candidate: Eigen::Map<MatrixType, MapOptions, StrideType>::Map(Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType, Eigen::Index, Eigen::Index, const StrideType&) [with PlainObjectType = const Eigen::Matrix<double, 6, 1, 0, 6, 1>; int MapOptions = 0; StrideType = Eigen::Stride<0, 0>; Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType = const double*; Eigen::Index = long int]
     inline Map(PointerArgType dataPtr, Index rows, Index cols, const StrideType
            ^
/usr/local/include/eigen3/Eigen/src/Core/Map.h:149:12: note:   candidate expects 4 arguments, 1 provided
/usr/local/include/eigen3/Eigen/src/Core/Map.h:135:12: note: candidate: Eigen::Map<MatrixType, MapOptions, StrideType>::Map(Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType, Eigen::Index, const StrideType&) [with PlainObjectType = const Eigen::Matrix<double, 6, 1, 0, 6, 1>; int MapOptions = 0; StrideType = Eigen::Stride<0, 0>; Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType = const double*; Eigen::Index = long int]
     inline Map(PointerArgType dataPtr, Index size, const StrideType& stride = S
            ^
/usr/local/include/eigen3/Eigen/src/Core/Map.h:135:12: note:   candidate expects 3 arguments, 1 provided
/usr/local/include/eigen3/Eigen/src/Core/Map.h:122:21: note: candidate: Eigen::Map<MatrixType, MapOptions, StrideType>::Map(Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType, const StrideType&) [with PlainObjectType = const Eigen::Matrix<double, 6, 1, 0, 6, 1>; int MapOptions = 0; StrideType = Eigen::Stride<0, 0>; Eigen::Map<MatrixType, MapOptions, StrideType>::PointerArgType = const double*]
     explicit inline Map(PointerArgType dataPtr, const StrideType& stride = Stri
                     ^
/usr/local/include/eigen3/Eigen/src/Core/Map.h:122:21: note:   no known conversion for argument 1 from ‘const int*’ to ‘Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >::PointerArgType {aka const double*}’
/usr/local/include/eigen3/Eigen/src/Core/Map.h:88:79: note: candidate: Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >::Map(const Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >&)
 mplate<typename PlainObjectType, int MapOptions, typename StrideType> class Map
                                                                             ^
/usr/local/include/eigen3/Eigen/src/Core/Map.h:88:79: note:   no known conversion for argument 1 from ‘const int*’ to ‘const Eigen::Map<const Eigen::Matrix<double, 6, 1, 0, 6, 1>, 0, Eigen::Stride<0, 0> >&’
src/openvslam/CMakeFiles/openvslam.dir/build.make:998: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/optimize/g2o/se3/shot_vertex.cc.o' failed
make[2]: *** [src/openvslam/CMakeFiles/openvslam.dir/optimize/g2o/se3/shot_vertex.cc.o] Error 1
CMakeFiles/Makefile2:138: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/all' failed
make[1]: *** [src/openvslam/CMakeFiles/openvslam.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Looking forward to your reply！

shinsumicco, 2019-05-28:
Could you confirm that you are using `Eigen 3.3.x`?

ZQxiaoshi, 2019-05-28:
I install eigen3.3.4 follow the documentation of OpenVSLAM:
    cd /path/to/working/dir
    wget -q http://bitbucket.org/eigen/eigen/get/3.3.4.tar.bz2
    tar xf 3.3.4.tar.bz2
    rm -rf 3.3.4.tar.bz2
    cd eigen-eigen-5a0156e40feb
    mkdir -p build && cd build
    cmake \
        -DCMAKE_BUILD_TYPE=Release \
        -DCMAKE_INSTALL_PREFIX=/usr/local \
        ..
    make -j
    make install
And I checked the eigen version and determined it was 3.3.4

ZQxiaoshi, 2019-05-30:
I slove the problem, by reship system.
But when I run the command：
“   # run localization
./run_video_localization -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_2/movie.mp4 -s ./ aist_living_lab_2/config.yaml --frame-skip 3 --no-sleep --map-db map.msg
”
... ...
Tracking Configuration:

[2019-05-30 18:55:38.916] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
[2019-05-30 18:55:39.489] [I] clear map database
[2019-05-30 18:55:39.489] [I] clear BoW database
[2019-05-30 18:55:39.489] [C] cannot load the file at map.msg
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot load the file at map.msg
*** Aborted at 1559213739 (unix time) try "date -d @1559213739" if you are using GNU date ***
PC: @     0x7f2c1c0bc428 gsignal
*** SIGABRT (@0x3e8000037ad) received by PID 14253 (TID 0x7f2c1f827bc0) from PID 14253; stack trace: ***
    @     0x7f2c1ec0c390 (unknown)
    @     0x7f2c1c0bc428 gsignal
    @     0x7f2c1c0be02a abort
    @     0x7f2c1c6f684d __gnu_cxx::__verbose_terminate_handler()
    @     0x7f2c1c6f46b6 (unknown)
    @     0x7f2c1c6f4701 std::terminate()
    @     0x7f2c1c6f4919 __cxa_throw
    @     0x7f2c1ef47f0d openvslam::io::map_database_io::load_message_pack()
    @     0x7f2c1eec6456 openvslam::system::load_map_database()
    @           0x41be78 mono_localization()
    @           0x41a877 main
    @     0x7f2c1c0a7830 __libc_start_main
    @           0x41ae09 _start
    @                0x0 (unknown)
已放弃 (核心已转储)

How to solve this problem？Looking forward to your reply！

beduffy, 2019-05-30:
I have the same problem as above:
```
[2019-05-30 20:24:42.049] [C] cannot load the file at map.msg
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot load the file at map.msg
```

shinsumicco, 2019-05-31:
Have you run `run_video_slam` with `--map-db map.msg` option before executing `run_video_localization`?
It seems that the map database (`map.msg`) does not exist at the specified path.

Please see [here](https://openvslam.readthedocs.io/en/master/simple_tutorial.html#tl-dr).

ZQxiaoshi, 2019-06-03:
I run this command successfully, thank you very much!

Shangareer, 2019-07-09:
I build the program according to the installation guide but have the same problem. Is there anyone who has a solution rather than reship the system?

Shangareer, 2019-07-09:
> I build the program according to the installation guide but have the same problem. Is there anyone who has a solution rather than reship the system?

Finally I solved this problem! For my case it was caused by 2 reasons.
1.It finds the eigen3 in directory "/usr/include/eigen3" in the "make -j" process, however the eigen is installed in the directory "/usr/local/include/eigen3". So I copied the directory "/usr/local/include/eigen3" to "/usr/include/".
2. The conflict between the g2o in ROS and g2o in system. I removed all the g2o and reinstalled  g2o in system(NOT in ROS) and then rebuilt this program successfully.

shaoxiang, 2019-07-17:
It worked for me.
sudo apt-get remove ros-kinetic-libg2o
then  cmake ......  make ......   again

closed 2019-09-11 by shinsumicco