xdspacelab/openvslam #36: pangolin docker error

gitunit, 2019-05-28:
im using nvidia-docker 2 and followed all the instructions.
the build was successful and the docker image starts but as soon as i want to try one of the mapping examples, i get the following error:

`error: XDG_RUNTIME_DIR not set in the environment.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
`

any idea what could be the problem?

shinsumicco, 2019-05-29:
Are you using `--runtime=nvidia` option for `docker run`?
Could you tell me about your host machine?

gitunit, 2019-05-29:
yes, i have used the runtime option. now i have tried on 2 different machines (both with NVIDIA, one has the Titan X and the other has dual graphics one NVIDIA GTX 970M, the other is intel on board but it is using solely NVIDIA in ubuntu).

here is the output on the machine with the Titan X (which is exactly the same error like on the other machine from yesterday):
`
error: XDG_RUNTIME_DIR not set in the environment.

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Unable to retrieve framebuffer options
*** Aborted at 1559121473 (unix time) try "date -d @1559121473" if you are using GNU date ***
PC: @     0x7f9f55111428 gsignal
*** SIGABRT (@0x10) received by PID 16 (TID 0x7f9f58f40bc0) from PID 16; stack trace: ***
    @     0x7f9f5831d390 (unknown)
    @     0x7f9f55111428 gsignal
    @     0x7f9f5511302a abort
    @     0x7f9f5597484d __gnu_cxx::__verbose_terminate_handler()
    @     0x7f9f559726b6 (unknown)
    @     0x7f9f55972701 std::terminate()
    @     0x7f9f55972919 __cxa_throw
    @     0x7f9f5624522b pangolin::ChooseFrameBuffer()
    @     0x7f9f56245c84 pangolin::CreateX11WindowAndBind()
    @     0x7f9f56247082 pangolin::RegisterX11WindowFactory()::X11WindowFactory::Open()
    @     0x7f9f56207ffc pangolin::CreateWindowAndBind()
    @     0x7f9f58bebac4 pangolin_viewer::viewer::run()
    @           0x41d435 mono_tracking()
    @           0x41bad2 main
    @     0x7f9f550fc830 __libc_start_main
    @           0x41c049 _start
    @                0x0 (unknown)
Aborted (core dumped)
`

shinsumicco, 2019-05-31:
Sorry for the late reply.

Umm... could you try `xeyes` and `glxgears` commands?

```
# in the docker container
apt update
apt upgrade -y
apt install x11-apps mesa-utils
# run xeyes
xeye
# run glxgears
glxgears
```

gitunit, 2019-06-03:
thx. that helped to debug the problem. xeyes worked but glxgears didn't. i got the following errors:
```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
```
then i found a solution [here](https://github.com/SoonminHwang/dockers/issues/1) which suggests to do the following:
```
docker run -it \
    -v /tmp/.X11-unix:/tmp/.X11-unix:rw \\
    --privileged \
    -e DISPLAY=$DISPLAY \
    -v /usr/lib/nvidia-384:/usr/lib/nvidia-384 \
    -v /usr/lib32/nvidia-384:/usr/lib32/nvidia-384 \
    --device /dev/dri \
    YOUR_DOCKER_IMAGE_NAME
In docker container,
```
in the docker container:
```
export PATH="/usr/lib/nvidia-384/bin":${PATH}
export LD_LIBRARY_PATH="/usr/lib/nvidia-384:/usr/lib32/nvidia-384":${LD_LIBRARY_PATH}
```

closed 2019-06-03 by gitunit