xdspacelab/openvslam #39: Docker build fails at the last two steps

Kelvinson, 2019-05-29:
Hi @shinsumicco thanks for your team's great work. I am so glad you are so considerate to publish  a docker version, which eases the amount of work to meet kinds of dependencies. 

When I built the docker image using nvidia-docker2, I came across problem at the last step, unlike issue [#11](https://github.com/xdspacelab/openvslam/issues/27)  which returned a non-zero code:2, the error here becomes " The command '/bin/sh -c set -x .... commands omitted' returned a non-zero code: 1. Do you have any idea what might cause this and how to fix it? Thank you very much!

[image]

Also the result of "docker image list" does not contain "openvslam-desktop"
[image]

shinsumicco, 2019-05-30:
Could you check `build/` is written in `.dockerignore`?

```
$ cd /path/to/openvslam
$ cat .dockerignore

...

# build directory
build/
cmake-build-debug/
cmake-build-release/
```

If not written, OpenVSLAM on your machine is not latest.
Please try the latest version :)

Kelvinson, 2019-05-30:
Thanks for your quick reply, actuallly the build directory(and debug release) is in the .gitignore file

shinsumicco, 2019-05-30:
Umm... could you remove `build/` directory, then execute `docker build` again?

```
$ cd /path/to/openvslam
$ rm -r build
$ docker build -t openvslam-desktop .
```

Kelvinson, 2019-05-30:
Hi thanks for  your reply, actually there is no build directory  in the openvslam directory since it was not built successfully.  from the  error " 
 cmake -DBUILD_WITH_MARCH_NATIVE=ON -DUSE_PANGOLIN_VIEWER=ON -DUSE_STACK_TRACE_LOGGER=ON -DBOW_FRAMEWORK=DBoW2 -DBUILD_TESTS=OFF ..
**CMake Error: The current CMakeCache.txt directory /openvslam/CMakeCache.txt is different than the directory /home/kelvinson/Documents/openvslam where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
The command "xxx " returned a non-zero code: 1**
" 
It seems there are path conflicts between my local CMakeCache.txt located in /home/xx/Documents/openvslam and the that in the docker image, which lcoates in the /openvslam/.

 The cachefile_dir in the CMakeCache.txt auto-generated by cmake  is follows,
`CMAKE_CACHEFILE_DIR:INTERNAL=/home/kelvinson/Documents/openvslam
`
So should I remove this line or edit its value to the directory of the CMakeCache.txt in the image( /openvslam/), or would you recommend any other solutions? Thank you very much!

Kelvinson, 2019-05-30:
Oh, I just removed the CMakeCache.txt and the CMakeOutput and rebuilt, it succeed, again thank you so much to open source such amazing work! Bravo!

closed 2019-05-30 by Kelvinson

shinsumicco, 2019-05-30:
Thank you for your report!
I'll add `CMakeCache.txt` and `CMakeOutput` to `.dockerignore`.
Thanks.