xdspacelab/openvslam #43: xdg runtime error and have to terminate manually

Kelvinson, 2019-05-30:
Hi, when I run the examples I came across following (maybe not) errors:

[image]

**_error: XDG_RUNTIME_DIR not set in the environment._**
[2019-05-30 11:19:22.978] [I] relocalization succeeded

(PangolinViewer: Frame Viewer:38): **_dbind-WARNING **: 11:19:22.997: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-JCbpe99zuj: Connection refused
Gtk-Message: 11:19:23.022: Failed to load module "canberra-gtk-module"
Gtk-Message: 11:19:23.024: Failed to load module "canberra-gtk-module"_**

It seems look like graphics problem.  I already run the
 
```
"xhost + local:" before run the docker 

using --runtime=nvidia and x11 forwarding 

```
Although it can run as in the tutorial, but I have to terminate it manually, or it would stop at the last frame, is that normal? Should it stop when all the video ends? 

Thank you for your amazing work!

shinsumicco, 2019-05-30:
> error: XDG_RUNTIME_DIR not set in the environment.

This warning occurs in our environment, too.
But I think it doesn't probably affect the behavior of the viewer.
Does the viewer launched in your environment work correctly?

> Although it can run as in the tutorial, but I have to terminate it manually, or it would stop at the last frame, is that normal? Should it stop when all the video ends?

Yes, it is normal behavior.
Could you try `--auto-term` option to terminate the viewer automatically?

Kelvinson, 2019-05-30:
Thanks for your reply, yes the viewer works correctly.  I will try that!

closed 2019-05-30 by Kelvinson