xdspacelab/openvslam #44: Segmentation fault around compute_orb_descriptor

CanCanZeng, 2019-05-30:
Hi @shinsumicco , thank you for sharing your work.
When I tried to run RGB-D dataset with openvslam, the program crashes often. Below is the error information:

```
*** Aborted at 1559231166 (unix time) try "date -d @1559231166" if you are using GNU date ***
PC: @     0x7efe96710ec8 openvslam::feature::orb_extractor::compute_orb_descriptor()
*** SIGSEGV (@0x7efd8bfffd47) received by PID 7995 (TID 0x7efe56d73700) from PID 18446744071763393863; stack trace: ***
    @     0x7efe944cb390 (unknown)
    @     0x7efe96710ec8 openvslam::feature::orb_extractor::compute_orb_descriptor()
    @     0x7efe967114bd openvslam::feature::orb_extractor::compute_orb_descriptors()
    @     0x7efe96713bcf openvslam::feature::orb_extractor::extract()
    @     0x7efe9670aa17 openvslam::data::frame::extract_orb()
    @     0x7efe9670cc7f openvslam::data::frame::frame()
    @     0x7efe96829671 openvslam::track::tracker::track_RGBD_image()
    @     0x7efe966a2454 openvslam::system::track_for_RGBD()
    @           0x41cd2d _ZZ13rgbd_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_SC_jbbbSC_ENKUlvE_clEv.constprop.773
    @     0x7efe941f0c80 (unknown)
    @     0x7efe944c16ba start_thread
    @     0x7efe93c5f41d clone
    @                0x0 (unknown)
Segmentation fault
```

I look into the function compute_orb_descriptor, and I suspect that the error is caused by memory access violation, because in the Macro `GET_VALUE(idx)` you did not check for image boundrays, I think there are chance accessing beyond borders unless you not detect FAST feature around image borders.

shinsumicco, 2019-05-31:
Thank you for your report.
We'll fix it soon. Could you wait for several days?

shinsumicco, 2019-06-02:
I couldn't reproduce the error you'd reported.
Could you post a sample image which causes SEGV?

CanCanZeng, 2019-06-02:
I use tum dataset reproduced the SEGV, the dataset is rgbd_dataset_freiburg3_long_office_household. the config file is shown below

```
# tum monocular model

#==============#
# Camera Model #
#==============#

Camera.name: "tum monocular"
Camera.setup: "monocular"
Camera.model: "perspective"

Camera.fx: 535.4
Camera.fy: 539.2
Camera.cx: 320.1
Camera.cy: 247.6

Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0
Camera.k3: 0.0

Camera.fps: 30.0
Camera.cols: 640
Camera.rows: 480

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 1000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.edge_threshold: 10
Feature.patch_size: 31

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 35

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400
```

and if I set mask parameter, the error is gone.
```
Feature.mask_rectangles:
  - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.9, 1.0]
  - [0.0, 0.1, 0.0, 1.0]
  - [0.9, 1.0, 0.0, 1.0]
```

CanCanZeng, 2019-06-04:
Hi @shinsumicco , have you reproduced the error on TUM dataset?

shinsumicco, 2019-06-04:
Sorry for late reply, @CanCanZeng. I'll check it tomorrow because I was busy.
Could you wait until then?

CanCanZeng, 2019-06-04:
Yeah, it is very kind of you to check it. Actually, I find that the function that computes orb descriptor is very similar to ORB_SLAM2, but ORB_SLAM2 will not come into this error, it  is so confusing. And I think this error do will happen when the key point is around image boarder, hope you can help me out.

shinsumicco, 2019-06-05:
I could reproduce the SEGV with the TUM dataset you'd reported.
The code of the ORB description is derived from [OpenCV's ORB extractor](https://github.com/opencv/opencv/blob/master/modules/features2d/src/orb.cpp#L239-L280), but OpenCV's doesn't cause any SEGV just like ORB-SLAM.
Thus, I think it is needed to filter keypoints around image boarders before describing ORB features.
I'm fixing it now. Thank you for your patience.

CanCanZeng, 2019-06-05:
Good news @shinsumicco , I find the cause. I set the parameter `Feature.edge_threshold: 10` inappropriately, it should be at least 16. Actuall I don't think this parameter should be exposed, since you already give the mask option.

shinsumicco, 2019-06-05:
Thanks. After all, `edge_threshold` and `patch_size` is fixed as `constexpr` variables for the reason you said.

Could you test [this PR](https://github.com/xdspacelab/openvslam/pull/50) with the `freiburg3_long_office_household` dataset?
In my environment, it does not cause any SEGV.

PR #50 (2019-06-08): Avoid SEGV when keypoints are detected near edges of images
close #44

- Fixed the two parameters: `Feature.edge_threshold` and `Feature.patch_size`
- Disabled padding of scaled images because it might cause SEGV in outside of ROI
closed 2019-06-19 by CanCanZeng