xdspacelab/openvslam #47: How to use mono-fisheye for this project?

ZQxiaoshi, 2019-06-03:
I see that fisheye is supported in openvslam project. My mono-fisheye video format and configuration are as follows, but it cannot be run. May I ask how to configure the file? Or could you provide an example for mono-fisheye?
[image]

THE ERROR AS BELOW：
$:  ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./mono_fisheye/movie.mp4 -s ./mono_fisheye/config.yaml --frame-skip 3 --no-sleep --map-db map.msg

[2019-06-03 14:33:18.922] [I] config file loaded: ./mono_fisheye/config.yaml
yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa

gitunit, 2019-06-04:
you need to provide a config file that has the distortions parameters, something like this:

```
# Tango 1 monocular model

#==============#
# Camera Model #
#==============#

Camera.name: "Tango 1 monocular"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 256.045
Camera.fy: 255.883
Camera.cx: 313.658
Camera.cy: 229.699

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0
Camera.k4: 0.0

Camera.fps: 30.0
Camera.cols: 640
Camera.rows: 480

Camera.color_order: "Gray"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 35

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400
```

with a config like above it will start without error but it will fail since the distortion params are all set to **0.0**

shinsumicco, 2019-06-04:
Thank you @gitunit !
For monocular fisheye camera models, please set at least the following options:

```
Camera.name: "camera name"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 250
Camera.fy: 250
Camera.cx: 320
Camera.cy: 240

Camera.k1: 0.0
Camera.k2: 0.0
Camera.k3: 0.0
Camera.k4: 0.0

Camera.fps: 30.0
Camera.cols: 640
Camera.rows: 480

Camera.color_order: "Gray"
```

`Camera.k1` - `Camera.k4` are distortion parameters in the form of [OpenCV's fisheye camera model](https://docs.opencv.org/trunk/db/d58/group__calib3d__fisheye.html).

ZQxiaoshi, 2019-06-10:
Thank you very much for your answer！@shinsumicco  @gitunit 
I calibrate my camera, and get the camera parameters.But the mono-fisheye SLAM  still doesn't work。In addition, I used data from the fish-eye data set on the Internet, which also did not work。There are two pictures below：
The first one is my fisheye camera and parameters,
[image]

The second is the fish-eye camera and parameters of the network data set:
[image]
As shown above，the mono-fisheye SLAM function cannot be realized.
Are there any open mono-fisheye datasets can I implement the SLAM function?

shinsumicco, 2019-06-11:
We've uploaded sample fisheye datasets [here](https://openvslam.readthedocs.io/en/develop/simple_tutorial.html#fisheye-datasets).
Would you try these datasets with the `develop` branch of OpenVSLAM?

I think distortion parameters of **TUM omnidirectional** dataset might be NOT in the form of [OpenCV's fisheye camera model](https://docs.opencv.org/trunk/db/d58/group__calib3d__fisheye.html).

ZQxiaoshi, 2019-06-13:
Thank you very much for sharing the mono-fisheye data set. I've successfully run the data set, It works very well, but I have one more question ,What is the accuracy of the mapping and positioning of this project? 
In addition, after calibrating the camera parameters, I can also run my own video. But it doesn't work very well，there are two problems  shown below:
First, Wrong Angle at the corner.It's a right Angle, but the result is not.
[image]

Second, Error calculating distance between key frames. 
[image]

It was good at first ,but when I went straight for about 200 meters, no distance can be calculated between keyframes.
How can I solve these problems so that they work as well as your data? Is it because my camera parameters are not accurate? 
I am looking forward to your reply！

Blackshaddock, 2019-07-10:
@ZQxiaoshi Do you use your own fisheye camera to run this project. i have the problem that it cannot be initialization.

letsdoitagain, 2019-07-30:
Hello,
is there any tool in ros to produce the calibration file given the required format? or one needs to write a custom script to produce this?

and what if i am not using a fish-eye model? what about a portable webcam (no noticable radial distortion on edges)?

tub-sgg, 2019-11-15:
hello, I can run the example.But when I try to use my own fisheye camera video, it does not work. I have changed the file .yaml.I am sure I do the right setting. But I want to know whether there is some limitation using fisheye cmaera??

YDSh510, 2020-03-12:
Hi,

How did you create fish eye model. If its opencv then how are you instantiating that model into your code