xdspacelab/openvslam #48: how to   save map  after  run  with video files?

lishuwei0424, 2019-06-03:
hi, first, thanks  for your amazing works, according to Simple Tutorial
  I run  this command for SLAM with Video Files.
./build/run_video_slam    -s   ./data/aist_living_lab_1/config.yaml    
                                        -m  ./data/aist_living_lab_1/movie.mp4    
                                         -v  ./orb_vocab/orb_vocab.dbow2  
                                         --frame-skip 3 \
                                        --map-db aist_living_lab_1_map.msg

when video is over, map is prbblem, Is it convenient to answer?

，thank you

shinsumicco, 2019-06-04:
Please click **Terminate** button to close the viewer and to save the map database.
Or, specify `--auto-term` option from command-line.

lishuwei0424, 2019-06-05:
thank you very much

closed 2019-06-07 by shinsumicco