xdspacelab/openvslam #54: Purpose of association paramter in TUM-Dataset example

ArashJavan, 2019-06-08:
Hello,
i am trying to get openvslam running on [TUM omnidirectional Dataset](https://vision.in.tum.de/data/datasets/omni-lsdslam), in commandline argument you should set the path to a association file. I would like to what exactly this file sohuld be?

Currently my yaml file for TUM Dataset looks as follow:

 `
# TUM monocular omnidirectional model for seq.00

#==============#
# Camera Model #
#==============#

Camera.name: "TUM monocular omnidirectional 00"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 442.843
Camera.fy: 442.66
Camera.cx: 320.65
Camera.cy: 253.91

Camera.k1: -0.0543194
Camera.k2: 0.24717305
Camera.k3: -0.0015186
Camera.k4: 0.00273411

Camera.fps: 50.0
Camera.cols: 640
Camera.rows: 512

Camera.color_order: "Gray"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 1000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.06
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.07
PangolinViewer.camera_line_width: 2
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -20
PangolinViewer.viewpoint_z: -0.1
PangolinViewer.viewpoint_f: 2800
`
Thanks for your help!

shinsumicco, 2019-06-08:
You may use `run_image_slam` for TUM monocular omnidirectional datasets instead of `run_tum_slam`.
Could you try it with `run_image_slam`?

shinsumicco, 2019-06-24:
@saweer kindly implemented the example for TUM RGBD dataset without association files at #61.
Please use it.

closed 2019-06-24 by shinsumicco

ArashJavan, 2019-07-05:
Thanks a lot for your response!