xdspacelab/openvslam #55: Trying to test several situations and several FOV.

renpoco, 2019-06-09:
Hi all.
First, Thank you for releasing and sharing this great engine.
I'm really impressed Visual SLAM worked so easily!
I want to do behavioral analysis from the video logs of cameras mounted on people, robots and pets in offline (and online if possible).
We are testing whether openvslam can hold feature points indoors and outdoors with some cameras, different FOVs.
I will share some of the results.

1.Garmin VIRB 360 x Outdoor
[image]
This condition was captured very fast.
And the movement trajectory was seemed correctly recognized.

2.Garmin VIRB 360 x Indoor
[image]
Unable to transition from INITIALIZE to MAPPING…
Perhaps because the camera was too close to the ceiling.

3.Sony FDR-X3000 x Outdoor
[image]
Unable to transition from INITIALIZE to MAPPING…
The condition may have been bad for shooting in a closed space.
Or because the angle of view was upward and flowing up and down?

4.Sony FDR-X3000 x Indoor
[image]
Unable to transition from INITIALIZE to MAPPING…
There is also a feature amount, and I do not know reason failed.

Q1.How do I avoid patterns where INITIALIZE does not proceed?
Is it not good at monotonous concrete etc?
Or should I stay as much as possible at first.
Or make amount of "skip" value smaller?
I do not understand ORB-Slam's habit itself so much, I am sorry.

Q2.If we can not judge whether it is fisheye or wide angle, like action cam including the FDR-X3000, which of the "fisheye" or the "perspective" would be work?
By the way, for the lens calibration value for FDR-X3000, the result of using opencv in other project was put into yaml and used as follows.

```yaml
#==============#
# Camera Model #
#==============#

Camera.name: "SONY FDR-X3000"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 814.9185791015625
Camera.fy: 815.2095336914062
Camera.cx: 971.536376953125
Camera.cy: 547.319091796875

Camera.k1: -0.08787094801664352
Camera.k2: 0.072215735912323
Camera.p1: 0.0006951461546123028
Camera.p2: -0.0006221338408067822
Camera.k3: -0.12616166472434998
Camera.k4: 0.09525316208600998
Camera.k5: 0.017126746475696564
Camera.k6: -0.12910930812358856

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 1080

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
# Xstart Xend Ystart Yend
#Feature.mask_rectangles:
#  - [0.0, 1.0, 0.7, 1.0]
#  - [0.45, 0.55, 0.3, 0.7]

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 35

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400
```

It would be appreciated if you could advise me about the part where usage is strange.
Thank you.

shinsumicco, 2019-06-12:
> Q1.How do I avoid patterns where INITIALIZE does not proceed?

Please capture scenes with rich textures and move the camera slowly for robust initialization.
Or, adjust [there parameters](https://github.com/xdspacelab/openvslam/blob/develop/src/openvslam/module/initializer.cc#L21-L23) by appending the entries to the config file. 

> Q2.If we can not judge whether it is fisheye or wide angle, like action cam including the FDR-X3000, which of the "fisheye" or the "perspective" would be work?

`"fisheye"` model is suitable for [the fisheye projection lens](https://wiki.panotools.org/Fisheye_Projection) due to the difference from the perspective projection.

I think `"perspective"` should be used for your FDR-X3000 because the distortion parameters you'd calibrated seem to be in the form of the OpenCV's perspective camera model.
If you want to use `"fisheye"` model, please calibrate the camera with the [OpenCV's fisheye camera model](https://docs.opencv.org/trunk/db/d58/group__calib3d__fisheye.html).
Please refer to [this](https://openvslam.readthedocs.io/en/develop/simple_tutorial.html#fisheye-datasets) as well.

renpoco, 2019-06-15:
Appreciate for your valuable advice, dear @shinsumicco !

I found missing of setting of yaml file.
It seems needs [Camera.color order: RGB]
*I really confused it because if set this to "RGB", preview image will be grayscale, and If set this to "Gray", preview will be colorful.

Anyway, it was successfully transitioned from INITIALIZE to MAPPING at last, but I could see the problem that the path was bent.

[image]
* This is a video of shooting up-direction while walking forward in the tunnel.

> "fisheye" model is suitable for the fisheye projection lens due to the difference from the perspective projection.
> I think "perspective" should be used for your FDR-X3000 because the distortion parameters you'd calibrated seem to be in the form of the OpenCV's perspective camera model.
> If you want to use "fisheye" model, please calibrate the camera with the OpenCV's fisheye camera model.
>Please refer to these examples as well.

As you pointed out, I performed calibration of FDR-X3000 with non fisheye (cvCalibrateCamera2).
However, specifying "perspective" will not display feature point markers at all ...

[image]

So I have no choice but to specify the "fisheye" mode, but it maight be the cause that the trajectory has been bent.
Could you release some example test set of "persepective(mono)" if possible?

I would like to do fish-eye calibration offcause, but unfortunately I can only use cvCalibrateCamera2 because my well-equipped OpenCV environment is C # (CVSharp) ...

And the following is my comment for other your advices.

> Please capture scenes with rich textures and move the camera slowly for robust initialization.
> Or, adjust there parameters by appending the entries to the config file.

I understood very well. I want to be careful when shooting images.
And I will try the parameter adjustment as well.

I have not tried the latest version you updated, so I will check it on the weekend.

renpoco, 2019-06-16:
When I recalibrated using OpenCV 4.10 & the latest version of CVSharp and run it on openvslam Ver 6/15, the curvature was improved a little.

[image]

However, using the parameters obtained with the normal Cv2.CalibrateCamera was looked better than using the "fisheye" calibration (Cv2.FishEye.Calibrate).

The rest seems to depend on the quality of the calibration, so I will close this subject.

Also, I attach a yaml file that reflects the latest calibration result of FDR-X3000.

[perspective]
``` yaml
# EuRoC monocular model

Initializer.num_ransac_iterations: 100
Initializer.num_min_triangulated_pts: 50
Initializer.parallax_deg_threshold: 1.0
Initializer.reprojection_error_threshold: 4.0
Initializer.num_ba_iterations: 20
Initializer.scaling_factor: 1.0

#==============#
# Camera Model #
#==============#

Camera.name: "SONY FDR-X3000"
Camera.setup: "monocular"
Camera.model: "perspective"

Camera.fx: 815.3360273810355
Camera.fy: 815.5596031971086
Camera.cx: 971.1646068807287
Camera.cy: 546.9207949596735

Camera.k1: 16.258414244295356
Camera.k2: 62.26015336479995
Camera.p1: 0.0006593265580827402
Camera.p2: -0.0007441284578242866
Camera.k3: 20.27963028367931
Camera.k4: 16.497211271364346
Camera.k5: 65.06457147215835
Camera.k6: 31.639523857087628

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 1080

Camera.color_order: "RGB"
#Camera.color_order: "Gray"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
# Xstart Xend Ystart Yend
#Feature.mask_rectangles:
#  - [0.0, 1.0, 0.7, 1.0]
#  - [0.45, 0.55, 0.3, 0.7]

#========================#
# Initializer Parameters #
#========================#

Initializer.scaling_factor: 5.0

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 35

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400

```

[fisheye]
```yaml
# EuRoC monocular model

Initializer.num_ransac_iterations: 100
Initializer.num_min_triangulated_pts: 50
Initializer.parallax_deg_threshold: 1.0
Initializer.reprojection_error_threshold: 4.0
Initializer.num_ba_iterations: 20
Initializer.scaling_factor: 1.0

#==============#
# Camera Model #
#==============#

Camera.name: "SONY FDR-X3000"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 648.83869149811119
Camera.fy: 658.97333601763785
Camera.cx: 964.33399280739184
Camera.cy: 539.95460650949985

Camera.k1: 1.1165908388693957
Camera.k2: -7.3726394897200844
Camera.k3: 10.517333316732358
Camera.k4: 4.0404909238482754

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 1080

Camera.color_order: "RGB"
#Camera.color_order: "Gray"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
# Xstart Xend Ystart Yend
#Feature.mask_rectangles:
#  - [0.0, 1.0, 0.7, 1.0]
#  - [0.45, 0.55, 0.3, 0.7]

#========================#
# Initializer Parameters #
#========================#

Initializer.scaling_factor: 5.0

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 35

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400

```

Thank you very much.

closed 2019-06-16 by renpoco

fangxu622, 2019-10-05:
hi, it's awesome , Cloud you  share you fisheye data，i want to test ，beause we are have Similar scenarios,and i alway fail to run my own dataset, thanks