xdspacelab/openvslam #57: some video is not track? loss

match08, 2019-06-14:
[hehuoren640.mp4.tar.gz](https://github.com/xdspacelab/openvslam/files/3288918/hehuoren640.mp4.tar.gz)
This video is track loss, but ORB-SLAM2 is track ok.

match08, 2019-06-14:
# Video monocular model

#==============#
# Camera Model #
#==============#

Camera.name: "Video monocular"
Camera.setup: "monocular"
Camera.model: "perspective" #perspective #equirectangular #fisheye

Camera.fx: 640
Camera.fy: 480
Camera.cx: 320
Camera.cy: 240

Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0
Camera.k3: 0

Camera.fps: 24.0
Camera.cols: 640
Camera.rows: 480

Camera.color_order: "RGB" #Gray

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 10000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 45

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400

ouyangandy, 2019-08-29:
did you fix it? @match08

match08, 2019-09-04:
> did you fix it? @match08

yes

ouyangandy, 2019-09-04:
> > did you fix it? @match08
> 
> yes

how？

ouyangandy, 2019-09-04:
> > > did you fix it? @match08
> > 
> > 
> > yes
> 
> how？

老兄，你咋解决的？

match08, 2019-09-04:
> > > > did you fix it? @match08
> > > 
> > > 
> > > yes
> > 
> > 
> > how？
> 
> 老兄，你咋解决的？

我还是使用了ORB SLAM2做了优化加入了线条特征, openvslam的你可以参考orb slam2.其实很像的。

closed 2019-09-11 by shinsumicco

ouyangandy, 2019-09-11:
@match08 能开源代码吗？