xdspacelab/openvslam #58: Camera pose format question

rauschaj, 2019-06-14:
Hello,

Thank you for your great work on this project! I am looking to extract the camera pose information and seem to have been able to extract the camera translation in world coordinates but have been unable to accurately extract the rotation.

I noticed that in `viewer.cc`, you are using the inverse of `cam_pose_cw`, as follows:
`const pangolin::OpenGlMatrix gl_cam_pose_wc(cam_pose_cw.inverse().eval());`

Does this mean that the camera rotation is inverted in this variable? The rotations (Euler angles) I was able to extract from `cam_pose_cw` were extremely small (generally <0.2) and did not align with what is seen in the Pangolin viewer.

Is something unusual about how the rotation matrix is being represented? I looked into the possibility that it was stored as a quaternion in (w, x, y, z) format instead of (x, y, z, w) and this seemed somewhat promising - is it possible this is what is going on?

Thank you!

shinsumicco, 2019-06-16:
Thank you for your interest in this project, @rauschaj.

> Does this mean that the camera rotation is inverted in this variable?

Yes, camera rotation and translation can be inverted with `inverse()` method.
More concretely, camera poses are represented in **4x4 matrices** (namely, [homogenous representation of SE3 group](http://www.xrpbot.org/wiki/SE3#The-homogenous-representation)), thus an inverse of camera pose matrix contains the inverted rotation matrix and translation vector.

You can extract rotation matrices and translation vectors in the world reference with `cam_pose_cw.block<3, 3>(0, 0)` (for rotation) and `cam_pose_cw.block<3, 1>(0, 3)` (for translation).

 Does it make sense?

shinsumicco, 2019-06-16:
Extraction of a rotation matrix and a translation vector from an SE3 matrix is written [here](https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/data/frame.cc#L138-L141).

The suffix `cw` means: a camera pose from **w**orld to **c**amera.

rauschaj, 2019-06-17:
Thank you for the very quick response, this is extremely helpful.

closed 2019-06-17 by rauschaj

santoshdahal2016, 2020-02-19:
Hello @shinsumicco ,  i got issue with  cam_pose_cw.block<3, 3>(0, 0) . the console shows as 

terminate called after throwing an instance of 'std::invalid_argument'
  what():  Error: Rotation Matrix does not describe a proper Rotation. There is scaling or a flipped axis involved. ( fabs(r.determinant() - 1) > tol ).

rahuldeo2047, 2020-04-06:
Ref only: https://github.com/xdspacelab/openvslam/issues/157

ryleymcc, 2020-12-11:
on pangolin, the camera translation is inverse, how do i fix this?