xdspacelab/openvslam #63: What is the accuracy of positioning and mapping？

ZQxiaoshi, 2019-06-19:
I can run my own monofisheye data on this algorithm. But it doesn't seem to work as well as the dataset. I want to know , what is the mapping and positioning accuracy of the openvslam? Is there a paper of the openvslam algorithm ?

shinsumicco, 2019-06-24:
Hmm, would you tell us some fisheye datasets which contain pose GT and calibration parameters?
We'll check the accuracy using them.

ZQxiaoshi, 2019-07-05:
Thank you very much for your reply！
Here are my mono-fisheye camera parameters:
`# Fisheye model for RICOH THETA S

#==============#
# Camera Model #
#==============#

Camera.name: "LUMIX DMC-GX8 8mm fisheye 1K"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.cols: 852
Camera.rows: 540
Camera.fps: 25.0

Camera.fx: 323.4671291042188
Camera.fy: 322.5352449399624
Camera.cx: 638.3589729623254
Camera.cy: 434.8604254403779

Camera.k1: 0.0710864
Camera.k2: 0.0384987
Camera.k3: -0.028936
Camera.k4: 0.00572445

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#========================#
# Initializer Parameters #
#========================#

Initializer.scaling_factor: 5.0
`
 But how do I get the "pose GT"? is it  the map.msg?
If I want to observe the actual size of SLAM's drawing output on your data set, how can I manipulate it to get the actual size of the output?

ZQxiaoshi, 2019-07-09:
In addition, I want to know the mapping and position accuracy of OpenVslam on your data set.

zainmehdi, 2019-12-17:
Hi, Thanks for the great work. I am also curious if it would be possible to get absolute position in world coordinate frame.

rahuldeo2047, 2020-04-06:
Ref only: https://github.com/xdspacelab/openvslam/issues/157

Zumbalamambo, 2020-05-28:
@rahuldeo2047 why are you referencing it to a question that's about rotation matrix?