xdspacelab/openvslam #67: How to display the camera pose of each frame at the front end of the web

jiqirenzhifu, 2019-06-25:
Hello, thank you very much for your great work. I am a newcomer to slam. I want to show the camera pose of each frame on the front end of the web to estimate how the camera moves to shoot video. Can you help me, or if you have any suggestions, any reply is very grateful. @

MikiyaShibuya, 2019-06-26:
Hi, could you open the viewer via any browser?
Please check the browser log [browser's developer tool(push F12 on browser) -> console].
It will say like " 139 KB   18.2 fps,   2282 pts,     22 kfs" in each line, then "22 kfs" means that there are 22 keyframes on the map.
If the number of keyframes is zero, the keyframes are not generated or transmitted.
Otherwise, the keyframes exist internally but not displayed on the viewer.

jiqirenzhifu, 2019-06-26:
@
Thank you very much for your reply, your answer is very helpful, I have solved the problem, good luck！

closed 2019-06-26 by jiqirenzhifu

rahuldeo2047, 2020-04-06:
@jiqirenzhifu : how was the issue solved. I am looking for similar solution. 

Also how did @MikiyaShibuya comment solves it. Optional.

Thank you
Ref only: https://github.com/xdspacelab/openvslam/issues/157