xdspacelab/openvslam #75: Run my own fisheye camera

Blackshaddock, 2019-07-10:
I am trying to get openvslam run my own fisheye camera, but it cannot be initializated, the fisheye video and configuration are as follows, could you tell me what to pay attention to when recording data?
[image]

[image]

Blackshaddock, 2019-07-11:
I have solved the problem

rahuldeo2047, 2019-09-03:
how?
What was the setup for the parameters?
Using OpenCV camera parameters?

ethanguo0327, 2019-09-24:
how do you fix that? @Blackshaddock

fangxu622, 2019-10-06:
怎么解决的的``我遇到同样的问题了 @Blackshaddock

Blackshaddock, 2019-10-10:
@fangxu622 你图片的分辨率太大了  提取特征点的时候 提取不到  你试试 把分别率降低点

Blackshaddock, 2019-10-10:
> how?
> What was the setup for the parameters?
> Using OpenCV camera parameters?

maybe you can use the function resize to reduce the image resolution

Blackshaddock, 2019-10-10:
@ethanguo0327 reduce the image resolution

closed 2019-10-10 by Blackshaddock

ethanguo0327, 2019-10-10:
> @ethanguo0327 reduce the image resolution

Thanks,I'll try