xdspacelab/openvslam #76: Localizing single image

nfoong, 2019-07-12:
Hi, thanks for this amazing project.

I have created a .msg map file using my own dataset (a 4K 360 RGB equirectangular video that I took around the corridors of my building using a RICOH THETA V) using `run_video_slam`. I was wondering if I took a new 360 RGB equirectangular photo somewhere in the space, is it possible for OpenVSLAM to localise this single image within the previously generated map?

From what I understand, `run_image_localization` requires a sequence of images for `track_for_monocular` to work. I'm asking if it's possible to perform localization with just a single image.

Thank you!

shinsumicco, 2019-07-15:
Thank you for your interest in this project.

You can use [relocalize()](https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/module/relocalizer.cc#L26) method of `relocalizer` class for single-frame relocalization.
`relocalize()` is called [here](https://github.com/xdspacelab/openvslam/blob/master/src/openvslam/tracking_module.cc#L296).

nfoong, 2019-07-15:
Hi, thank you for your response.

I have 2 videos, `video_1` and `video_2`. I generated a map database using `video_1` and saved it as `map.msg`.

I then loaded `map.msg` and tried using the `relocalize()` method (`run_image_localization` does this for me) to localize a single frame in `video_2`. It is unable to localize this new frame, and some debugging reveals that this is because the variable `num_matches = 2` when `min_num_bow_matches_ = 20`, and so the candidate frame is discarded.

I then run `run_video_localization` on `video_2` and `map.msg` with the `--mapping` flag turned on, and modified the code so that it saves the new updated map database as `map_updated.msg`. When I load `map_updated.msg` and use `relocalize()` on the same `video_2` frame again, this time the localization succeeds. 

The screenshot below shows the path taken in `video_2` as a bold line and the path taken in `video_1` as a thin line. As seen from the screenshot, the camera pose (indicated by the triangle) in the frame I am trying to localize is not far from the path taken by `video_1` (the thin line). In real life, it is about a 1meter displacement.

[image]

Hence I wonder why OpenVSLAM is unable to localize this frame in `map.msg` even though it is in the same corridor with only a slight displacement in position? I see in the `aist_living_lab` tutorial that there were no issues at all when localizing `aist_living_lab_2` using the map generated from `aist_living_lab_1`. Could it be because the corridors in `aist_living_lab` are tighter with more features and/or multiple loops were performed when generating the map from `aist_living_lab_1`? Or was it that when recording `aist_living_lab_2`, special care was taken to walk in exactly the same line along the corridors as `aist_living_lab_1`?

In any case, I've attached the two frames of the same corridor from `video_1` and `video_2` below, respectively. I hope to hear from you soon. Thank you for your time!

[image]
[image]

JinglingZhu, 2019-07-25:
Hi @nfoong ,
Thanks for your sharing. I am also trying to localize a single image. May I know how you do `relocalize()` by using method `run_image_localization`. Do you mean you extract a frame from `video_2` first and put the frame image into a dir, then pass to the parameter `image_dir_path`? Would you please share the code you extract a frame from a video?

Thanks!

alexs7, 2019-12-06:
@nfoong Hello ! Same here! I am trying to relocalise a single image. I was wondering if the image was taken from another device that should still work as long as I pass the correct config file ?