xdspacelab/openvslam #77: Prebuilt windows binaries? Issues with SuiteSparse BLAS

Jason420Git, 2019-07-12:
Hi, I would really like to give this library a go to see how it is. However, I have been having a really big trouble getting suitesparse to find BLAS inside the cmake. I seem to have tried so many variations but I really cannot seem to find BLAS. 

Does anyone have a pre-built version of this for windows? (ideally MSVC 2015) but any would be really great to try. Thanks!

antithing, 2019-09-05:
Hi, I always use this:

https://github.com/jlblancoc/suitesparse-metis-for-windows

easy to build with cmake. :)

I am struggling to get openvslam built on windows, were you able to do it?

Jason420Git, 2019-09-08:
I tried that one but also had problems. After a few weeks of trying to solve all the g2o errors I gave up. I instead installed WLS (see the tutorial) so I could install it under Ubuntu terminal whilst still being on Windows. I was able to get that working within a day.

closed 2019-09-08 by Jason420Git

shinsumicco, 2020-04-26:
@Jason420Git 
Would you try this?
https://github.com/shinsumicco/openvslam-on-msvc/