xdspacelab/openvslam #78: insta360 one X camera model

skypu3, 2019-07-16:
I can't find the insta one X model config, can someone provide me the camera parameters?

shinsumicco, 2019-07-17:
Please change the image size (`Camera.cols` and `Camera.rows`) of the configuration for THETA S according to the image resolution of insta360 One X.

choward232, 2019-07-23:
Should the lens on this camera be classified as fisheye or equirectangular?

shinsumicco, 2019-09-11:
`equirectangular` model is suitable for insta360 one X.

EnriqueSolarte, 2019-10-17:
Dear all,  
Thank for your contribution. 
@shinsumicco, for equirectangular projection, the estimation of the essential matrix considers spherical projection, isn't it? 
Tracing the code, I saw you are using the 8-point algorithm for such estimation. How does this 8-p implementation consider a different projection such as equirectangular (spherical projection)? 
In the paper, there are no details about this. Do you know where this description is available?
In advance thank you for your response.

jungladicitta, 2020-01-21:
@skypu3 @choward232 
how are you streaming image raw from insta360 onex to openvslam?