xdspacelab/openvslam #79: How to get camera model on iPhone XS mounted fisheye lens

skypu3, 2019-07-17:
Hi team,

I want to find pin hole camera model on iPhone XS mounted with fisheye lens
So I get "intrinsic matrix" to find the fx, fy, cx, cy from below:
https://developer.apple.com/documentation/avfoundation/avcameracalibrationdata/2881135-intrinsicmatrix
And I use the camera-calibration-ios github to find the distortion_coefficients.
https://github.com/thorikawa/camera-calibration-ios
Here is the camera model I find:
#==============#
# Camera Model #
#==============#

Camera.name: "iPhone"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 6190.428
Camera.fy: 6190.428
Camera.cx: 1493.424
Camera.cy: 1972.364

Camera.k1: -3.2828323353358818e-01
Camera.k2: 7.8798773365756081e-02
Camera.k3: -3.0927226354753820e-03
Camera.k4: 8.6781907034207198e-03

Camera.fps: 30.0
Camera.cols: 900
Camera.rows: 1600

However, we can't get openvslam video slam work normally, does anyone have know how to setup the camera model on iOS ?

shinsumicco, 2019-07-18:
Could you set `Camera.cols` and `Camera.rows` properly in the configuration?

skypu3, 2019-07-24:
@shinsumicco, Camera.cols and Camera.rows are the video resolution, right ?

lgotkowski, 2020-05-06:
Hi, 
did you get openVSLAM run on an IPhone, with localization? 
If so, could you briefly mention you toolset and pipeline how you managed to get it run on ios?

Best regards
Lukas

andrewcccc, 2020-07-13:
Hi @skypu3, did you just use this example: https://github.com/thorikawa/camera-calibration-ios. I tried to run the app but the app does not show those values as you indicated.

mirellameelo, 2020-07-13:
@andrewcccc 

I did that way > https://github.com/xdspacelab/openvslam/issues/368#issuecomment-649209592

(It was not with iPhone, but the source doesn't matter since all you need is to get some frames/pictures).

andrewcccc, 2020-07-13:
Hi @mirellameelo. Thanks for the response! I am trying that with an iPad. However, I tried to run the app but the output on Xcode just showing "not found". I am not too sure how to get the camera_parameter.yml with all these camera values. Do you have a moment to explain? 

Best regards,

mirellameelo, 2020-07-13:
@andrewcccc 
Sure. You can find a description of what I did in the comment https://github.com/xdspacelab/openvslam/issues/368#issuecomment-649209592

Let me know if I wasn't clear on any of these steps.

andrewcccc, 2020-07-13:
@mirellameelo, Thanks for the response! I will dig in a bit more. But is it applied to Swift as well for the example code? The example codes indicated for python and C++ only?

mirellameelo, 2020-07-13:
@andrewcccc You want to get your camera parameters and then build your .yalm file, am I right?
If yes, print a chessboard, make a video showing the chessboard while making movements around it, always showing the entire board. Later, bring the video into your pc. Is it possible? If yes, with the OpenCV library, read this video and save some frames of it. 15 - 20 frames from different positions should be enough. OBS: you can make it using pictures once the pictures have the same resolution of the video. But for what I remember, Apple is too restrict for camera configurations...so, using the video is safer.

Now, with the frames/pictures of the chessboard in different positions, you can follow the tutorial I mentioned and get the parameters. Or you need this to be inside this Swift? If yes, then I have no idea how to make it, sorry.

ma2b0043, 2020-10-21:
@mirellameelo, I tried to do the same task using an iphone 6 camera. I recorded a video converted them into images and when I ran the code provided at https://www.learnopencv.com/camera-calibration-using-opencv/ , cv2.findChessboardCorners returned false and null value, which basically means that the code couldnt detect the image correctly. I dont know either my dataset for this task is not suitable for this code or something else isnt working, can you help me out ?

I'm attaching a rgb colored image in my dataset. I've even converted them to graysacle image still the cv2.findChessboardCorners return false and null value.
[image]

ma2b0043, 2020-10-21:
@mirellameelo and also i wanted to ask you is there a way to automatically generate a config. yaml file by using a normal video of 3d space not a 2d object like this checkerboard? at runtime?