xdspacelab/openvslam #81: VR180 stereo tracking with Vuze XR

nfoong, 2019-07-19:
Hi,

I've been trying to get stereo tracking to work on a Vuze XR VR180 camera, which has 2 X 210⁰ fisheye lenses. The 4K video that comes out from the camera is 3840x1920 in resolution. I then use OpenCV to split each frame in half and store the left and right halves in separate directories. I tried calibrating the camera using a chessboard and OpenCV and tried using a modified version of run_kitti_slam, but the video would not track.

One 1920x1920 frame from the left camera is shown below:

[image]

I'm confused about which camera model I should use (fisheye, perspective, equirectangular), and about how I should perform calibration for this camera to get the intrinsic parameters (fx, fy, cx, cy, etc.). From the frame above it looks like when OpenCV reads the frame some transformation is already performed on it, stretching out the top and bottom such that it is like "equirectangular", but from your code I understand that stereo tracking does not support equirectangular.

How then, should I get stereo tracking to work on input frames like this?

Thank you.

shinsumicco, 2019-07-22:
Sorry for the late reply, @nfoong.

It seems that `perspective` camera model will match to your Vuze XR VR180 camera.
In our system, stereo image pairs must be inputted as stereo-rectified ones.
Would you input stereo-rectified images to the system by utilizing [cv::fisheye::initUndistortRectifyMap](https://docs.opencv.org/3.4/db/d58/group__calib3d__fisheye.html#ga0d37b45f780b32f63ed19c21aa9fd333)?

If input stereo image pairs are undistorted and stereo-rectified, our system behaves the same way between `perspective` and `fisheye` models.

Please refer to [a sample code for EuRoC stereo sequence](https://github.com/xdspacelab/openvslam/blob/master/example/run_euroc_slam.cc#L149)?

nfoong, 2019-07-24:
Hi @shinsumicco, thank you for your reply.

I suppose that if input stereo pairs are undistorted and stereo-rectified, then in the configuration file all the distortion coefficients `(k1,k2,p1,p2,k3)` should be set to zero, but intrinsic parameters `(fx,fy,cx,cy)` should remain the same?

Also I noticed in the EuRoc configuration file, there is a parameter called `Camera.focal_x_baseline`. Is this the distance between the two cameras? If so, what units is this in? Thanks!

JinglingZhu, 2019-08-02:
I have the similar question, I have a ZED stereo camera and it can output undistorted and rectified images. So does it means the distortion coefficients and  intrinsic parameters have no affect in the SLAM running? Another question is that ZED provided a calibration file with two cameras' parameters (both left camera and right camera have a fx, fy, cx, cy). What should I use in yaml fie for openvslam?

nfoong, 2019-08-12:
Hi @shinsumicco, sorry to bother you again.

I've used `cv::fisheye::initUndistortRectifyMap` to undistort and stereo-rectify images as the input, just like in `run_euroc_slam`. The frame displayed by `Pangolin` during slam seems about right. However, it still loses tracking after a few frames, and the point cloud map generated seems incorrect (the points corresponding to the bottom edge of the wall should be straight but in the generated point cloud it appears to be slanted, see image below).

[image]

I suspect that it has something to do with an incorrect value of `Camera.focal_x_baseline`, as I think the rest of the values are calibrated correctly. May I know how to obtain the value of `Camera.focal_x_baseline`? Is it just the distance between the two cameras in millimeters?

The distortion coefficients in the config file have been set to 0, like in the EuRoc config example. Are there any other config parameters that I should take note of? Any other general suggestions about what could be wrong would also be greatly appreciated.

Thank you.

[image]