xdspacelab/openvslam #82: Issue running the first video_slam example on Ubuntu

choward232, 2019-07-20:
After installing all the dependencies and installing openvslam in accordance to the installation guidelines, I attempted to run the sample project but the command ./run_video_slam (followed by all the appropriate flags) caused a sigabrt and my core to dump. Here is the stack trace:

*** Aborted at 1563580445 (unix time) try "date -d @1563580445" if you are using GNU date ***
PC: @     0x7f31fc8d2e97 gsignal
*** SIGABRT (@0x3e800000ebd) received by PID 3773 (TID 0x7f31cde18700) from PID 3773; stack trace: ***
    @     0x7f31ff4a0890 (unknown)
    @     0x7f31fc8d2e97 gsignal
    @     0x7f31fc8d4801 abort
    @     0x7f31fc91d897 (unknown)
    @     0x7f31fc92490a (unknown)
    @     0x7f31fc92be75 cfree
    @     0x7f31fb9f5068 g2o::OptimizableGraph::~OptimizableGraph()
    @     0x7f31ff86a0f2 openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7f31ff7f1801 openvslam::module::initializer::create_map_for_monocular()
    @     0x7f31ff7f24c3 openvslam::module::initializer::initialize()
    @     0x7f31ff761a6c openvslam::tracking_module::initialize()
    @     0x7f31ff7644d5 openvslam::tracking_module::track()
    @     0x7f31ff764852 openvslam::tracking_module::track_monocular_image()
    @     0x7f31ff755d3c openvslam::system::track_for_monocular()
    @     0x55d678f83b85 (unknown)
    @     0x7f31fcf5a66f (unknown)
    @     0x7f31ff4956db start_thread
    @     0x7f31fc9b588f clone
Aborted (core dumped)

Any tips on how to fix this?

shinsumicco, 2019-07-22:
Would you run `cmake` for g2o and OpenVSLAM with `-DBUILD_WITH_MARCH_NATIVE=OFF` option?
Please check [here](https://openvslam.readthedocs.io/en/master/trouble_shooting.html).

choward232, 2019-07-22:
Tried both and still no luck!

choward232, 2019-07-22:
The compiler warns me about not finding Qt5, is that also a dependency I need to install?

BrauSolano, 2020-02-10:
I am having the same error choward232 had