xdspacelab/openvslam #83: Run error with my fisheye camera

tkym525, 2019-07-22:
I am trying to get openvslam run my own fisheye camera, but map cannot be initializated or miss feature tracking right after intialization. I think the accuracy of camera calibration is bad. Checking configuration as follows, the value of "min x" is positive and "max x" is negative. Does it happen in usual case?

[image]

shinsumicco, 2019-07-22:
It is unusual behavior.
`max_x` should be greater than `min_x`. (Also `max_y` should be greater than `min_y`.)

I think `cy` seems inaccurate because `cy` is usually half of the `rows`.
Would you try `cy=480.0` for debugging?

nfoong, 2019-07-22:
Hi, I achieve a similar result as @tkym525 when trying my own 210⁰ fisheye camera. Max x and max y are negative, and the map cannot be initialized:

[image]

I've done manual calibration which seems to work fine. Using openCV's `initUndistortRectifyMap` I get the following before correction (left) and after correction (right):

[image]

Given that the calibration seems to be correct, any idea why tracking doesn't work? Thanks!

nfoong, 2019-07-23:
@shinsumicco I did the calibration again and somehow managed to fix the negative max x issue. But the tracker still fails to initialise and the viewer keeps giving this kind of output:

[image]

Here is the configuration used:

[image]

Any idea why the tracker fails to initialise?

tkym525, 2019-07-25:
@shinsumicco Thank you. I finally achieved SLAM.  First, I did the calibration again and adjusted the value manually. Negative max x issue is still remain with the distortion(k1, k2, k3, k4). Second, I used "run_image_slam" instead of  "run_video_slam". I'm investigating the cause that "run_video_slam" doesn't work.

@nfoong I see the image seems to be a little blurry and there are few feature points. I think you should slow down the moving speed and do an experiment in more busy places.