xdspacelab/openvslam #84: video_localization does not load back previously saved map

szabolcsvelkei, 2019-07-23:
- openvslam built on ubuntu16.04 (as a windows 10 WSL)
- all tutorials and slam localization examples are working perfectly
- run_video_slam can generate nice maps based on our drone video (monocular/perspective)
- "small" maps are working with run_video_localization (soft definition of "small": not so many keyframes/landmarks/loops (?)   for example: 300 keyframes, 15k landmarks and 1-3 loops is good)
- "larger" maps could not be loaded by run_video_localization

Dump:
>  Camera Configuration:
> - name: Genius UWA
> - setup: Monocular
> - fps: 30
> - cols: 640
> - rows: 360
> - color: RGB
> - model: Perspective
>   - fx: 490.16
>   - fy: 489.273
>   - cx: 315.847
>   - cy: 181.171
>   - k1: 0.226793
>   - k2: -1.00307
>   - p1: -0.00314689
>   - p2: -0.003051
>   - k3: 1.41029
>   - min x: 13.4129
>   - max x: 627.711
>   - min y: 8.6343
>   - max y: 353.583
> ORB Configuration:
> - number of keypoints: 1000
> - scale factor: 1.2
> - number of levels: 8
> - initial fast threshold: 20
> - minimum fast threshold: 7
> Tracking Configuration:
> 
> [2019-07-23 16:32:01.452] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
> [2019-07-23 16:32:02.380] [D] CONSTRUCT: data::camera_database
> [2019-07-23 16:32:02.381] [D] CONSTRUCT: data::map_database
> [2019-07-23 16:32:02.383] [D] CONSTRUCT: data::bow_database
> [2019-07-23 16:32:02.390] [D] CONSTRUCT: publish::frame_publisher
> [2019-07-23 16:32:02.390] [D] CONSTRUCT: publish::map_publisher
> [2019-07-23 16:32:02.394] [D] CONSTRUCT: module::initializer
> [2019-07-23 16:32:02.398] [D] CONSTRUCT: module::relocalizer
> [2019-07-23 16:32:02.399] [D] CONSTRUCT: tracking_module
> [2019-07-23 16:32:02.400] [D] CONSTRUCT: mapping_module
> [2019-07-23 16:32:02.401] [D] CONSTRUCT: global_optimization_module
> [2019-07-23 16:32:02.404] [I] clear map database
> [2019-07-23 16:32:02.405] [I] clear BoW database
> [2019-07-23 16:32:02.424] [I] load the MessagePack file of database from map.msg
> [2019-07-23 16:32:06.186] [I] decoding 1 camera(s) to load
> [2019-07-23 16:32:06.187] [I] load the tracking camera "DJI Mavic Air" from JSON
> [2019-07-23 16:32:06.716] [I] decoding 456 keyframes to load
> [2019-07-23 16:32:13.368] [I] decoding 16324 landmarks to load
> [2019-07-23 16:32:13.418] [I] registering essential graph
> terminate called after throwing an instance of 'std::out_of_range'
>   what():  _Map_base::at
> *** Aborted at 1563892333 (unix time) try "date -d @1563892333" if you are using GNU date ***
> PC: @     0x7fa59b765428 gsignal
> *** SIGABRT (@0x26a) received by PID 618 (TID 0x7fa59e920840) from PID 618; stack trace: ***
>     @     0x7fa59c0c1390 (unknown)
>     @     0x7fa59b765428 gsignal
>     @     0x7fa59b76702a abort
>     @     0x7fa59bdaf84d __gnu_cxx::__verbose_terminate_handler()
>     @     0x7fa59bdad6b6 (unknown)
>     @     0x7fa59bdad701 std::terminate()
>     @     0x7fa59bdad919 __cxa_throw
>     @     0x7fa59bdd62cf std::__throw_out_of_range()
>     @     0x7fa59dfe70ce openvslam::data::map_database::register_graph()
>     @     0x7fa59dfe90a5 openvslam::data::map_database::from_json()
>     @     0x7fa59e00d09c openvslam::io::map_database_io::load_message_pack()
>     @     0x7fa59df8f126 openvslam::system::load_map_database()
>     @           0x41b668 mono_localization()
>     @           0x419e57 main
>     @     0x7fa59b750830 __libc_start_main
>     @           0x41a3e9 _start
>     @                0x0 (unknown)
> Aborted (core dumped)

Does anyone have know what is the problem?

szabolcsvelkei, 2019-07-23:
Update: the same issue appears with the  aist_living_lab_3  sample dataset

jarrodhaas, 2019-07-24:
interesting -- just ran a test with run_video_localization on a map containing 1102 keyframes and 58051 landmarks, no problems. 

If we run into any issues, I'll check back in.

szabolcsvelkei, 2019-07-24:
It's not crystal clear yet but it seems that the problem exists if there are keyframes unrelated to the rest of the graph. (They not connected with lines when you show the graph in the viewer.)
I raised the limit of the number of ORB keypoints upto 2-3k  and the map is actually loadable.

I checked the aist_living_lab_3 dataset again with default parameters and it has the same problem at the end of the video: the very last keyframes are unconnected.

jarrodhaas, 2019-07-24:
ok thanks for providing that info! Again, we'll keep an eye on it and check back if we learn anything, as we'll be testing pretty heavily over the next couple of months.

shinsumicco, 2019-07-24:
Thank you for the report, @szabolcsvelkei and @jarrodhaas .
We'll investigate the cause and fix it within a few weeks.
I would appreciate your cooperation!

jarrodhaas, 2019-07-30:
I've been able to use run_localization to import a map with > 90,000 landmarks and > 2300 keyframes, with no issues. everything seems to be saving and loading fine on our end with map areas of this size.

We're running on macosx 10.14.5, and this is our config file:

Camera.name: "Vuze XR"
Camera.setup: "monocular"
Camera.model: "equirectangular"

Camera.fps: 30.0
Camera.cols: 1920
Camera.rows: 960

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles:
  # - [0.0, 1.0, 0.0, 0.1]
  - [0.0, 1.0, 0.70, 1.0]

shinsumicco, 2019-07-30:
@szabolcsvelkei
Could you try to save and load a map with the version at [my forked repository](https://github.com/shinsumicco/openvslam/tree/fix/map-load-segv)?
Please use `fix/map-load-segv` branch.

shinsumicco, 2019-11-05:
@szabolcsvelkei
Did you solved?

PR #88 (2019-08-21): Avoid SEGV when a pre-build map is loaded
fix #84