xdspacelab/openvslam #87: Accessing or extracting point cloud data directly or through the written database file

KaushikPulagara, 2019-07-26:
Hello.

I was working with video file data sets and was wondering if there is a way to access the point cloud data directly? Which then could be used to generate a mesh out of.

MikiyaShibuya, 2019-07-27:
Hi,
The generated map-database(*.msg) contains the point cloud data and so on.
These data are compressed by MessagePack.
Try to use the following Python script so that extract point cloud as csv.
Please install "msgpack" previously. 
Thanks,
```
"""
Copyright (c) 2019,
National Institute of Advanced Industrial Science and Technology (AIST),
All rights reserved.

This work is licensed under the terms of the 2-Clause BSD license.
For a copy, see <https://github.com/xdspacelab/openvslam/blob/master/LICENSE>.
"""

import sys
import msgpack
import json

def main(bin_fn, dest_fn):

    # Read file as binary and unpack data using MessagePack library
    with open(bin_fn, "rb") as f:
        data = msgpack.unpackb(f.read(), use_list=False, raw=False)

    # The point data is tagged "landmarks"
    landmarks = data["landmarks"]
    print("Point cloud has {} points.".format(len(landmarks)))

    # Write point coordinates into file, one point for one line
    with open(dest_fn, "w") as f:
        for id, point in landmarks.items():
            pos = point["pos_w"]
            f.write("{}, {}, {}\n".format(pos[0], pos[1], pos[2]))
    print("Finished")

if __name__ == "__main__":
    argv = sys.argv

    if len(argv) < 3:
        print("Unpack all landmarks in the map file and dump into a csv file")
        print("Each line represents position of landmark like \"x, y, z\"")
        print("Usage: ")
        print("    python pointcloud_unpacker.py [map file] [csv destination]")

    else:
        bin_fn = argv[1]
        dest_fn = argv[2]
        main(bin_fn, dest_fn)
```

shinsumicco, 2019-08-02:
Does it work for you, @KaushikPulagara ?

closed 2019-08-20 by shinsumicco

claudiolanfranco, 2020-01-23:
From the python files:  # The point data is tagged "landmarks"
which other data are presents in the database .msg file format created ?
are present the 'graph' list ?

belal-ibrahim, 2020-06-25:
May i know how was the result?