xdspacelab/openvslam #92: Access to the coordination of key frames

ArmastusChen, 2019-08-12:
I have read the issue #87 , which allows me to extract the detailed position of landmarks. But I would like to get the accurate position of key_frames of a map. I have checked the value "trans_cw" in the map.msg, but it looks inconsistent with the result displayed in the provided viewer. Besides, the real distance between any two points is needed in my case.

Could anyone please help me to solve it? Thank you so much!

ArmastusChen, 2019-08-15:
I have solved this problem, but coming up with another one... How could I get the image related to each key frame?

closed 2019-08-15 by ArmastusChen

Wsupatgiat, 2019-10-08:
Hey, I would like to know how did you extract the keyframe positions. Thanks in advance.

ArmastusChen, 2019-10-09:
[map_location_extractor.py.zip](https://github.com/xdspacelab/openvslam/files/3705269/map_location_extractor.py.zip)
You could try code in this file (simply change the input and output path). Hope it will be helpful.

Wsupatgiat, 2019-10-09:
Thanks a ton! :D

t-martyniuk, 2019-10-15:
why do you use incremented fps value?
`            fps = int(vidcap.get(cv2.CAP_PROP_FPS)) + 1
`
you then multiply it with point['ts'] and it returns non-integer values. Is it on purpose?

monschine, 2020-02-02:
Thanks @ArmastusChen for your extraction code, I have some issues with RigidTransform, as doing a simple install of the autolab_core seems not to include it. Can you tell me how to install it without a ROS environment:
https://berkeleyautomation.github.io/autolab_core/install/install.html#option-2-install-from-source-for-ros

Thanks in advance

rahuldeo2047, 2020-04-06:
Ref only: https://github.com/xdspacelab/openvslam/issues/157

OmarLaz97, 2020-04-10:
@ArmastusChen thanks a lot for the provided code, did you find a way to get the image corresponding to each keyframe ? And if there is a way to know the corresponding keyPoints for every keyFrame ?

ArmastusChen, 2020-04-10:
> Thanks @ArmastusChen for your extraction code, I have some issues with RigidTransform, as doing a simple install of the autolab_core seems not to include it. Can you tell me how to install it [...]

I am sorry for the late reply. I installed it without ROS env on my Mac. I just simply follow the official instruction to install it (I choose the Protobuf for UI since I failed to install SocketViewer).

ArmastusChen, 2020-04-10:
> why do you use incremented fps value?
> ` fps = int(vidcap.get(cv2.CAP_PROP_FPS)) + 1`
> you then multiply it with point['ts'] and it returns non-integer values. Is it on purpose?
> @ArmastusChen

I am sorry for the late reply. Yes, you can run the code and see if you want to modify it.

ArmastusChen, 2020-04-10:
> @ArmastusChen thanks a lot for the provided code, did you find a way to get the image corresponding to each keyframe ? And if there is a way to know the corresponding keyPoints for every keyFrame ?

Q1: Yes, you can check the code carefully. I saved the image for each keyframe.
Q2: Yes, please see line 82 in the code. Check the dictionary (variable 'point'), and you can get what you want.
Please let me know if you have any questions.

OmarLaz97, 2020-04-10:
> Q1: Yes, you can check the code carefully. I saved the image for each keyframe.
> Q2: Yes, please see line 82 in the code. Check the dictionary (variable 'point'), and you can get what you want. [...]

Thank you so much for your fast reply !! 
I did take a look on the variable "point" and found what i want there as you said. Thank you again.

OmarLaz97, 2020-04-25:
@ArmastusChen greetings again, i was trying to implement navigation using 2D projection of the 3D map and using your method for getting the position of each keyframe, but i am facing an issue that the position of the landmarks when plotted in 2D and the position of the keyframes when plotted too do not match. The keyframes are shifted. Any idea what could be causing this issue ?
[image] i attached a picture for my issue.

ArmastusChen, 2020-04-25:
> @ArmastusChen greetings again, i was trying to implement navigation using 2D projection of the 3D map and using your method for getting the position of each keyframe, but i am facing an issue that [...]

Hi. From my knowledge, the landmarks are some feature points of keyframes. For example, the edge point of an object in your scene should be a landmark, whose position is different from your camera position. So it should be shifted. Did you run the .map file with GUI provided? All the yellow points (or red? if I remember correctly) should be the landmarks.

OmarLaz97, 2020-04-25:
Yes indeed i saw these colored points while building the 3D sparse map. What i think that the value of the key "pos_w" in the map data is in a different coordinate space than the keyframes position that we got from multiplying the "rot_cw" matrix and "trans_cw" vector. So my question is how can i get both to be on the same coordinate space ?

ArmastusChen, 2020-04-26:
> Yes indeed i saw these colored points while building the 3D sparse map. What i think that the value of the key "pos_w" in the map data is in a different coordinate space than the keyframes position [...]

Hi. I am sorry that I don't have any idea about it. I just used the openvslam as a small part of my project so I didn't delve into it. By the way, please post your solution if you work out it. Thank you!

OmarLaz97, 2020-04-26:
@ArmastusChen  My main task was to perform localization on a 2D grid using VSLAM, so i used your technique and added  _-np.transpose_ to get the position of every keyframe in world coordinates.

  ```
trans_cw = np.asarray([trans_x, trans_y, trans_z])
rot_cw = np.asarray([rot_cw_x, rot_cw_y, rot_cw_z, rot_cw_w])
rigid_cw = RigidTransform(rot_cw, trans_cw)
pos = np.matmul(-np.transpose(rigid_cw.rotation), trans_cw)
```
But the positions of the keyframes used while mapping and localization were off, and they were not drawn over the 2D grid map, that i have drawn using the key _pos_w_ for the landmarks from the _map.msg_ data.

This was the output while using the mapping keyframes (the black dots are the keyframes).
[image]

I then tried to output the camera center for every localization keyframe from the _openvslam_ file _run_image_localization.cc_ using this piece of code 

```
const openvslam::Mat44_t  framedata = SLAM.feed_monocular_frame(img, frame.timestamp_, mask);
const openvslam::Vec3_t&  trans_cw = framedata.block<3, 1>(0, 3);
const openvslam::Mat33_t& rot_cw = framedata.block<3, 3>(0, 0);

const openvslam::Mat33_t& rot_wc = rot_cw.transpose();
const openvslam::Vec3_t& cam_center_ = -rot_wc * trans_cw;
```
and also added this in the _keyframe.cc_ inside the _to_json()_ code.

```
const openvslam::Vec3_t&  trans_cw = cam_pose_cw_.block<3, 1>(0, 3);
const openvslam::Mat33_t& rot_cw = cam_pose_cw_.block<3, 3>(0, 0);

const openvslam::Mat33_t& rot_wc = rot_cw.transpose();
const openvslam::Vec3_t& cam_center_ = -rot_wc * trans_cw;
```
and added the _cam_center_ to the json output using this.

`{"cam_center", convert_translation_to_json(cam_center_)}`

Both of the code blocks use the same techniques as the code provided by you, but for some reason that i did not quite get the keyframes position was accurate and drawn over the map when we used the _cam_center__ data.

This is the output of drawing the mapping keyframes.
[image]

And this is the output of drawing the localization keyframes
[image]

I don't know what is causing this behavior, if you have any idea please tell me.

zlrpuck, 2020-05-07:
@OmarLaz97 Why did you transpose the rotation matrix before pose calculation?

OmarLaz97, 2020-05-07:
@zlrpuck I need to get the inverse of _rot_cw_ and rotation matrices are orthogonal therefore the inverse is the transpose. Please refer to this link for more explanation. [https://math.stackexchange.com/questions/82602/how-to-find-camera-position-and-rotation-from-a-4x4-matrix](url)

OmarLaz97, 2020-05-16:
@ArmastusChen Do you have any idea how can i get the direction where the camera is looking ?

chyohoo, 2020-05-16:
> @ArmastusChen Do you have any idea how can i get the direction where the camera is looking ?

direction = rot_wc.block<3,1>(0,2)

mirellameelo, 2020-05-16:
@OmarLaz97 Hi! I'm trying to do just the same thing, an occupancy grid map from the point clouds. Do you reproject every single landmark to the 2D plane and then dilate those? Do you mind detailing more?
I'm trying to use Bresenham's line algorithm to draw a line between the keyframe position until all the keymaps visible by that KF. Do you have any idea how can I get only the landmarks visible? I'm not sure if it is the "local_landmark" variable.

Tks!

OmarLaz97, 2020-05-16:
@chyohoo Thank you for your response, i was trying the output of your answer but it seems like i am not able to draw the direction vector correctly on the 2D map using the X and Z values. I am using the center of the camera as the origin and the X and Z values to get the slope of the line i want to draw (The direction of the camera) . But still the direction is wrong. Do you have any idea why this is happening ?

OmarLaz97, 2020-05-16:
@mirellameelo You can find the landmarks in the "landmarks" key when you unpack the map.msg file. To generate the 2D map, I used the same technique provided here https://github.com/skylook/ORB_SLAM2-gridmap, the code responsible for the map generation is in pointCloudToGridMap2D.py, you may find it useful.

mirellameelo, 2020-05-16:
@OmarLaz97 Thanks for the answer!
I'm trying to replicate this code for Openvslam too! I unpacked the .msg file and accessed these map points, but what the pointClouldToGridMad2D.py use as input is (a) the keyframe trajectory (save_keyframe_trajectory function in Openvslam and SaveKeyFrameTrajectoryTUM in ORBSLAM) and (b) the map points with its respective timestamp (SaveWithTimestamps function in ORBSLAM but I couldn't find the respective in Openvslam). Trying to figure it out...
What did you exactly use as input then? Tks!

OmarLaz97, 2020-05-16:
@mirellameelo What i used as input:
From "landmarks" i used "pos_w" and "1st_keyfrm" to get the position and the corresponding key frame for the landmark.
From "keyframes" i used "cam_center", please refer to this comment https://github.com/xdspacelab/openvslam/issues/92#issuecomment-619566227 for more explanation about this key.

OmarLaz97, 2020-05-23:
@mirellameelo i had the same results using the demo video to produce the corresponding 2D map. i think the map is too sparse because the original 3D map was big with a low number of key points. The map i showed here was much smaller with a higher density of key points.

monschine, 2020-06-03:
@mirellameelo I'm also trying to plot all the values into a 2d context. My approach is to use the frame_trajectory.txt generated with --eval-log flag. 

the file is giving a list of camera positions like this :
0.1 0.0167 0.00211 0.066 -0.002 0.00980 0.00496 0.9999

Can anyone confirm this values are, following this order ?
**keyframe_id, x, y, z, q1, q2, q3, w**

And my question would be how to extract the value of yaw rotation from the quaternion ?
I have tested some approach, but it seems I do not get the correct values regarding the yaw extraction. 
[image]

https://threejs.org/docs/#api/en/math/Quaternion

Thanks in advance

monschine, 2020-06-05:
Thanks for the answer @mirellameelo !

Zumbalamambo, 2020-06-05:
@mirellameelo could you please suggest what modifications that I need to do in openvslam to get the grid map as yours?

mirellameelo, 2020-06-08:
@Zumbalamambo, Hi! Here in this comment (https://github.com/xdspacelab/openvslam/issues/92#issuecomment-632885757), I describe what I did. 

To be more specific, in trajectory_io.cc (A) I modified the save_keyframe_trajectory() function to also print the keyframe id (keyfrm->id_), and (B) I modified the save_json_file() function to also output the first keyframe which sees that landmark (point->first_keyfrm_id_). With that, I was able to read these .txt files, and by associating the keyframe_id, lines could be defined. 

Please, feel free to check it out the code in my git, branch ros2.