xdspacelab/openvslam #94: What spherical camera can be used for live SLAM

hugefeng, 2019-08-19:
I want to use OpenVLSAM with a Spherical camera for indoor mapping. I bought an Insta 360 OneX , but cannot obtain the live video stream.  I have to record the scene in advance, and use the video for SLAM.  May I ask what type of spherical camera can be used for live SLAM?

shinsumicco, 2019-08-20:
We use THETA V and THETA Z1 for live streaming via [USB Video Class](https://en.wikipedia.org/wiki/USB_video_device_class).

Unfortunately, a driver for UVC 1.5, which is used for video streaming in THETA V and Z1, is not provided for Linux, thus we use macOS and Windows for live vSLAM.

Alternatively, you can use [Wi-Fi transfer mode](http://theta360.guide/community-document/live-streaming.html#_theta_v_streaming_with_wi_fi_span_class_icon_i_class_fa_fa_puzzle_piece_fa_2x_i_span) of THETA V and Z1 for Linux.

Please see also below:
http://theta360.guide/community-document/live-streaming.html#_theta_v_streaming_with_wi_fi_span_class_icon_i_class_fa_fa_puzzle_piece_fa_2x_i_span
https://discourse.ros.org/t/cameras-with-360-fov-in-ros/1833
https://qiita.com/dcm_kitade/items/6ff790fbb0a97913265b (Japanese)

hugefeng, 2019-08-20:
Thank you very much! I will try it.

closed 2019-09-27 by shinsumicco

duguowei1000, 2020-01-22:
Hi, i am thinking of using THETA S to perform real-time SLAM on ubuntu.  It does live stream at 1080p only. WIll the lower resolution affect the feature extraction?