xdspacelab/openvslam #97: Serialized writing or reading database

CanCanZeng, 2019-08-21:
The writing and reading database now is transforming type and writing or reading at once, when the database is rather large, say, I created a map with 2000 keyframes and 20K landmarks, writing or reading stage uses all of my 16GB memory. For larger dataset I think the memory usage is unaccptable, maybe serialized writing and reading can alleviate this problem?

shinsumicco, 2019-09-03:
Thank you for your suggestion, @CanCanZeng.

I think it is difficult to directory apply serialized writing and reading to OpenVSLAM because pose-graph optimization and local/global BA alters the map database continuously.
One solution is to use SQlite3 for a global map, then read/write a local map from/to it on demand.

CanCanZeng, 2019-09-05:
I think SQlite3 would be a good solution, since the Struct from Motion algorithm COLMAP also use SQlite3 to manage database, the database of COLMAP is also very complex and large, but SQlite3 handles very well.