xdspacelab/openvslam #101: Cholesky failure when running dataset

CanCanZeng, 2019-09-02:
Hi @shinsumicco , sorry to bother you again. In order to get more accurate map, I force the tracking thread to wait for the optimize process of local mapping module to complete, and raise the local mapping module's optimze iteration number at the same time, the code is as follows:

```
while (!mapper_->get_keyframe_acceptability()) {
            std::this_thread::sleep_for(std::chrono::microseconds(1000));
}

```
inserted in here:
https://github.com/xdspacelab/openvslam/blob/f1e90c704a2f11b1708057e8369c5805ccc828ca/src/openvslam/tracking_module.cc#L173

and chage the `num_first_iter_` to 50 and `num_second_iter_` to 100

then I run video_slam and tum_rgb_slam in monocular mode, both run into this warning:
```
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
int g2o::csparse_extension::cs_cholsolsymb(const cs_di*, number_t*, const cs_dis*, number_t*, int*): cholesky failed!
```
and the map's scale chages violently, sometimes cause the mapping model divergence.

I don't understand, is there some bug in g2o? but if I change `num_first_iter_` and `num_second_iter_` to default values, the warnning is gone.
Can you check this situation? You can use TUM's rgbd_dataset_freiburg3_long_office_household_validation dataset or other video dataset to reproduce this situation.

shinsumicco, 2019-09-03:
Thank you for posting, @CanCanZeng .

It seems that over-optimization causes scale-drift when the number of fixed keyframes is insufficient.
I will check it within a few weeks.

riematrix, 2019-10-29:
Hi,
We encountered the same problem but the result is more severe. Keyframe were suddenly adjusted to far away locations as soon as this message shows up. I haven't reproduce this for now and after I reproduce it I can submit more messages if you need.
Thanks.

shinsumicco, 2019-11-05:
@riematrix 
Could you explain it in more detail?

riematrix, 2019-11-05:
@shinsumicco

CanCanZeng, 2020-05-08:
Hi, I find that the error is caused by csparse, if I change to g2o::LinearSolverEigen, the error is gone.

Have you tested the efficiency and robusty of g2o::LinearSolverEigen and g2o::LinearSolverCSparse? @shinsumicco 

And I find that for large problem, the g2o::LinearSolverPCG will be much more efficient, do you think it is a good idea do change the solver? @shinsumicco

SYSUHPK, 2020-07-21:
Hello, I have the same problem like you. Can you explain the solution with more details? Like where to change   g2o::LinearSolverCSparse to g2o::LinearSolverEigen @CanCanZeng

CanCanZeng, 2020-07-26:
@SYSUHPK 
https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/src/openvslam/optimize/local_bundle_adjuster.cc#L101
https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/src/openvslam/optimize/global_bundle_adjuster.cc#L34
https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/src/openvslam/optimize/graph_optimizer.cc#L29

SYSUHPK, 2020-08-03:
Thanks a lot!!!