xdspacelab/openvslam #102: build issue while building openvslam

rahuldeo2047, 2019-09-03:
Followed: https://openvslam.readthedocs.io/en/master/installation.html

There is no build error(s) at any other step(s) before this.

I get following error message in step "When building with support for SocketViewer".

```
$ protoc --version
 libprotoc 2.6.1

$ pkg-config --cflags protobuf 
 -pthread

# I do have openCV installed with ROS-kinetic

$ pkg-config --cflags opencv
 Package opencv was not found in the pkg-config search path.
 Perhaps you should add the directory containing `opencv.pc'
 to the PKG_CONFIG_PATH environment variable
 No package 'opencv' found
```

Error msgs

`map_segment.proto:1:10: Unrecognized syntax identifier "proto3".  This parser only recognizes "proto2".
src/socket_publisher/CMakeFiles/socket_publisher.dir/build.make:62: recipe for target 'src/socket_publisher/map_segment.pb.cc' failed
make[2]: *** [src/socket_publisher/map_segment.pb.cc] Error 1
CMakeFiles/Makefile2:448: recipe for target 'src/socket_publisher/CMakeFiles/socket_publisher.dir/all' failed
make[1]: *** [src/socket_publisher/CMakeFiles/socket_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
`

rahuldeo2047, 2019-09-03:
The Pangolin build is working great.

shinsumicco, 2019-09-03:
It seems that the version of `protobuf` and `protoc` you use are too old.
Could you use the version 3.x of them, which can parse `proto3` schemes?

If you use Ubuntu 16.04, it is needed to build the newest protobuf from source.

```
wget -q https://github.com/google/protobuf/archive/v3.6.1.tar.gz
tar xf v3.6.1.tar.gz
cd protobuf-3.6.1
./autogen.sh
./configure \
    --prefix=/usr/local \
    --enable-static=no
make -j4
make install
```

rahuldeo2047, 2019-09-03:
Hey, thank you for the reply.

I tried that and still protoc --version gives version 2 as mentioned in first comment.

shinsumicco, 2019-09-03:
Dis you install `protoc` via apt?

Please try this then rebuild and reinstall protoc from souce as mentioned in my comment:
```
apt purge libprotobuf-dev protobuf-compiler
```

rahuldeo2047, 2019-09-03:
ok will try that

rahuldeo2047, 2019-09-03:
ref: https://github.com/xdspacelab/openvslam/issues/32

shinsumicco, 2019-11-05:
@rahuldeo2047
Did you solved?

rahuldeo2047, 2019-11-05:
I had to shift to Ubuntu 18.04 .
Now it is building and running fine.

shinsumicco, 2019-11-05:
Thank you for the report. @rahuldeo2047

closed 2019-11-05 by shinsumicco