xdspacelab/openvslam #104: Any guideline for any camera setup?

rahuldeo2047, 2019-09-03:
Thanks for this repo.

I am looking for the guideline for setting up any camera whatsoever. Is there any such hardware limitation?
say 

- a mobile phone's video without having a need for an extra lens for fisheye etc.
- a webcam in a similar fashion

I guess the process can be

- get a camera
- determine camera parameter (at what distance?, spec?, opencv)
- capture and preprocess the video as per the params.
- make a config accordingly
- run the video slam

Let me know for the clarity if need be.

Thanks again for bringing such a great slam algo up to the common developers (ppl).

rahuldeo2047, 2019-09-05:
Ref: https://github.com/raulmur/ORB_SLAM2/issues/332

pedrobagatin, 2019-09-10:
Hi, I am trying to do the same. I obtained "uvc camera" (webcam) configuration file by doing [monocular calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) after saving a yaml file is generated (however a little different from the examples)

Check also in their website there is a section for uvc camera (examples).

fangxu622, 2019-09-10:
I have the same question, how to get camera models params, opencv or else? cloud you give some refecrence?

pedrobagatin, 2019-09-10:
Also, did some experiments using uvc camera and got "track lost" problems  when turning right or left on the corners. As you can see, example videos are recorded using 360 degrees camera.

pedrobagatin, 2019-09-10:
If you go to: openvslam/src/openvslam/camera/ you can see all the parameters you need to put in the config.yaml (there are three types of camera: perspective, equirectangular and fisheye).

For example (perspective):

perspective(const std::string& name, const setup_type_t& setup_type, const color_order_t& color_order,
                const unsigned int cols, const unsigned int rows, const double fps,
                const double fx, const double fy, const double cx, const double cy,
                const double k1, const double k2, const double p1, const double p2, const double k3,
                const double focal_x_baseline = 0.0);

So I think in the config.yaml we need to specify: fx,fy,cx,cy,k1,k2,p1,p2 and k3

fangxu622, 2019-09-10:
hmm, Thanks all. Which camera model should select for fisheye(FOV=220, panorama) like this

[image]

I select the opencv fisheye model for get camera matrix.

shinsumicco, 2019-09-11:
Thank you for posting, @rahuldeo2047.

The camera parameters should be estimated via pre-calibration.
The parameters used in OpenVSLAM is in the same fashion with the ones in OpenCV.
(`k1`, `k2`, `p1`, `p2`, and `k3` for perspective cameras, `k1`, `k2`, `k3`, and `k4` for fisheye cameras.)

You can calibrate a camera with OpenCV's methods and a proper checker or grid-pattern board.

shinsumicco, 2019-09-11:
The fisheye camera model seems good for the image you posted, @fangxu622.

rahuldeo2047, 2020-04-06:
For reference : https://github.com/xdspacelab/openvslam/issues/157

MikeWangWZHL, 2020-07-28:
How to  implement a dual fisheye camera model? Is there any more specific guidance?