xdspacelab/openvslam #105: make error at ubuntu16.04

IAMGGKK, 2019-09-04:
Hi,

I install all dependencies follow document.

I encounter some problem when run "make",here is the wrong message:

[ 80%] Built target test_openvslam_solve_essential_solver
[ 82%] Built target test_openvslam_util_trigonometric
/home/ggkk/openvslam-master/src/pangolin_viewer/viewer.cc: In member function ‘pangolin::OpenGlMatrix pangolin_viewer::viewer::get_current_cam_pose()’:
/home/ggkk/openvslam-master/src/pangolin_viewer/viewer.cc:213:60: error: no matching function for call to ‘pangolin::OpenGlMatrix::OpenGlMatrix(const Eigen::Matrix<double, 4, 4>&)’
     const pangolin::OpenGlMatrix gl_cam_pose_wc(cam_pose_cw);
                                                            ^
In file included from /usr/local/include/pangolin/display/viewport.h:31:0,
                 from /usr/local/include/pangolin/gl/gl.h:30,
                 from /usr/local/include/pangolin/pangolin.h:33,
                 from /home/ggkk/openvslam-master/src/pangolin_viewer/viewer.h:11,
                 from /home/ggkk/openvslam-master/src/pangolin_viewer/viewer.cc:1:
/usr/local/include/pangolin/display/opengl_render_state.h:301:8: note: candidate: pangolin::OpenGlMatrix::OpenGlMatrix()
 inline OpenGlMatrix::OpenGlMatrix() {
        ^
/usr/local/include/pangolin/display/opengl_render_state.h:301:8: note:   candidate expects 0 arguments, 1 provided

I have tried some solutions, but it still doesn't work.

Can someone help me with this?

rahuldeo2047, 2019-09-04:
Let us make sure we have followed: https://openvslam.readthedocs.io/en/master/installation.html completely.
And not a single step has been missed.
Ref: https://github.com/xdspacelab/openvslam/issues/102

shinsumicco, 2019-09-11:
Could you uninstall and re-install Pangolin according to the documentation, @IAMGGKK?

IAMGGKK, 2019-09-15:
I have solve this problem by re-install. Thank you very much!

closed 2019-11-05 by shinsumicco