xdspacelab/openvslam #108: Windows building problems

antithing, 2019-09-05:
I have all dependencies built and linked, but every time i try to build (windows 10, visual studio 2017)

It compiles for a few minutes, then visual studio freezes, locks up and starts spinning a wheel until I crash it.

Has anyone successfully built this code on windows?

Thanks!

shinsumicco, 2019-09-11:
Thank you for posting, @antithing.

We are currently working on the build on Windows.
Could you please wait until then?

We are very grateful if you post here any instructions you did for the build with Visual Studio.

antithing, 2019-09-28:
Hi @shinsumicco, do you have an estimate for releasing the first windows version? 
I am happy to test for you if you need!

Thanks again.

schweitzer, 2019-10-10:
Hello,

I've experimented the same issues (and many others) on msvc 2019.

This seems the same as in #28, but the I don't think that the mentioned workaround is a real fix.

For now I don't understand why msvc find multiple definitions in optimise/g2o module but I will keep trying, and I let you know if I find something interesting.

Furthermore to have a full windows 10 support you also need to remove (or to convert) all CXX_FLAGS in the main CMakeLists.txt, along with removing OpenMP support (msvc is stuck to version 2.0).

rgolovanov, 2019-10-15:
Same linking errors. Did anybody fix it?

antithing, 2019-10-17:
I tried for a while but could not get to the bottom of the g2o errors. I have never seen these issues with g2o before.

If anyone finds the fix please post it here!

rgolovanov, 2019-10-17:
These errors mean multiple instantiation of template classes with same arguments (types) which are happened at g2o side and OpenVSLAM as well. Only one temporary workaround I found is to forcibly suppress these errors by adding linker argument `add_link_options(/FORCE:MULTIPLE)` but I've not tested library well yet, so be attentive.

shinsumicco, 2019-10-18:
Dear all,
Sorry for the late reply, @antithing.
Temporarily I don't have access to Windows environment, so I cannot try to build OpenVSLAM on Windows at the moment.
Does the build problem seem to be caused by the multiple instantiation of template classes?

rgolovanov, 2019-10-18:
@shinsumicco most probably the root cause is in multiple template instantiation with same template arguments. Let's see, e.g. `Error	LNK2005	"public: virtual void __cdecl g2o::BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1> >::setMeasurement(class Eigen::Matrix<double,2,1,0,2,1> const &)" 
(?setMeasurement@?$BaseEdge@$01V?$Matrix@N$01$00$0A@$01$00@Eigen@@@g2o@@UEAAXAEBV?$Matrix@N$01$00$0A@$01$00@Eigen@@@Z) 
already defined in perspective_pose_opt_edge.obj`

There are two instantiation of `g2o::BaseEdge<2,class Eigen::Matrix<double,2,1,0,2,1>>`:
* In **OpenVSLAM** (_...\openvslam\src\openvslam\optimize\g2o\se3\perspective_pose_opt_edge.h_):
```class mono_perspective_pose_opt_edge final : public ::g2o::BaseUnaryEdge<2, Vec2_t, shot_vertex>```
* and in **g2o** (g2o\types\slam2d\edge_xy_prior.h):
``` class G2O_TYPES_SLAM2D_API EdgeXYPrior : public BaseUnaryEdge<2, Vector2, VertexPointXY>```

Which leads to instantiation of same `BaseEdge` class:
`class BaseUnaryEdge : public BaseEdge<D,E>`

Absolutely same thing for other linking errors. You can find instantiation of template classes with equal template arguments.

antithing, 2019-10-18:
I can confirm that by making the following changes the lib will build and run on windows 10, visual studio 2017.

#define _USE_MATH_DEFINES
#include <cmath>
 or #define M_PI in a few headers.

add include path for \SuiteSparse_config.h
add include path for g2o_config.h
added lapack and blas libs as dependencies

I also swapped the LinearSolverCSparse for LinearSolverCholmod, as i had some unresolved errors in CSparse.

However, using my own live stereo camera at 640 x 360, I see a frame time of around 110 ms. Much slower than it should be based on the paper.  Is there anything I can look at to speed this up?

Thanks!

reunanen, 2020-01-02:
For the record: I, too, got OpenVSLAM built and running on Windows 10, Visual Studio 2017. In addition to the discussion above, found some very useful information [here](https://qiita.com/na0ki_ikeda/items/2310dc8dd70713de98e1) (in Japanese, but Google translation helps).

It did take quite a while to complete all the steps, though. Moreover, I'm not a big fan of CMake generated project files that have absolute paths all over. Therefore, I wanted to try to figure out a cleaner way of setting up a Visual Studio build, and eventually came up with [this](https://github.com/reunanen/openvslam-build-vs). So, if anyone wants to test this alternative approach, here are the instructions:

1. `git clone https://github.com/reunanen/openvslam-build-vs`
2. `cd openvslam-build-vs`
3. `git submodule update --init --recursive`
4. Open `openvslam.sln` (in repo root) in Visual Studio 2017
5. Build
6. Test as described [here](https://openvslam.readthedocs.io/en/master/simple_tutorial.html)

Out of the box, it'll expect that your processor has [AVX2](https://en.wikipedia.org/wiki/Advanced_Vector_Extensions#Advanced_Vector_Extensions_2). (This requirement is possible to relax, though – if anybody needs to run this on lower-end CPUs, please let me know.)

At any rate, the runtime speed seems to be ok, at least on my machine:
[image]

Also, [all tests complete successfully](https://ci.appveyor.com/project/reunanen/openvslam-build-vs/builds/29852302#L2834).

Oh, and thanks a lot for this library, it looks really great!

shinsumicco, 2020-04-26:
@antithing 
Would you try this?
https://github.com/shinsumicco/openvslam-on-msvc/

closed 2020-04-26 by shinsumicco