xdspacelab/openvslam #110: How to obtain exactly config.yaml?

pedrobagatin, 2019-09-10:
I did [monocular calibration](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration) however generated config.yaml is a little different from the examples:

os.txt
```
# oST version 5.0 parameters

[image]

width
640

height
480

[narrow_stereo]

camera matrix
1523.306037 0.000000 146.852037
0.000000 1380.330890 116.819146
0.000000 0.000000 1.000000

distortion
0.853147 0.038534 -0.060017 -0.116773 0.000000

rectification
1.000000 0.000000 0.000000
0.000000 1.000000 0.000000
0.000000 0.000000 1.000000

projection
1556.893188 0.000000 130.445498 0.000000
0.000000 1453.644043 109.946910 0.000000
0.000000 0.000000 1.000000 0.000000
```

And .yaml file:

```
image_width: 640
image_height: 480
camera_name: narrow_stereo
camera_matrix:
  rows: 3
  cols: 3
  data: [1523.306037, 0.000000, 146.852037, 0.000000, 1380.330890, 116.819146, 0.000000, 0.000000, 1.000000]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [0.853147, 0.038534, -0.060017, -0.116773, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000]
projection_matrix:
  rows: 3
  cols: 4
  data: [1556.893188, 0.000000, 130.445498, 0.000000, 0.000000, 1453.644043, 109.946910, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000]
```

shinsumicco, 2019-09-11:
Thank you for posting, @pedrobagatin.
You need to reconfigure the config file according to the style as shown in the examples.
Could you try this one?

```
Camera.name: "test name"
Camera.setup: "monocular"
Camera.model: "perspective"

Camera.fx: 1523.306037
Camera.fy: 1380.330890
Camera.cx: 146.852037
Camera.cy: 116.819146

Camera.k1: 0.853147
Camera.k2: 0.038534
Camera.p1: -0.060017
Camera.p2: -0.116773
Camera.k3: 0.000000

Camera.fps: 20.0  # please reconfigure according to your camera
Camera.cols: 640
Camera.rows: 480

Camera.color_order: "Gray"  # please reconfigure according to your camera
```

However, the camera calibration you did seems to have much error because `cx` is far from the half of `fx`. (It is the same for `cy` and `fy`.)

rhidra, 2021-02-11:
Hi,
Could we add more documentation on how to create this config.yaml file ?
I can't find anywhere what does the `model`, `fx`, `cx`, `k1` or `p1` variables correspond to.