xdspacelab/openvslam #114: Skipping frames when saving trajectory to a file

joannatarko, 2019-09-13:
Hi!

When I run video slam with option --eval-log, it skips frames when saving to the frame_trajectory.txt file. The number of skipped frames varies depending on the length of the video. For example, it's 8 frames for 720 frames long video and 10 frames for 1000 frames long one. The corresponding file with tracking times contains times for every frame. I set --frame-skip value to 1 and don't use --no-sleep option.

Is it possible to get all camera trajectory saved to a file without any frames skipped? I'm new to OpenVSLAM and maybe I don't fully understand the influence of all parameters.

I'm running OpenVSLAM on Docker with SocketViewer. This is the command I use:
`./../run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./video.mp4 -c ./configCam.yaml --frame-skip 1 --eval-log --map-db ./video_map.msg`

shinsumicco, 2019-11-05:
@joannatarko 
A frame trajectory only after initialization is written in `frame_trajectory.txt` because initialization process is essential to track camera poses.
If you want to record all the trajectory, one solution is to use a stereo or an RGBD camera.
It is more easy for them to initialize the system than monocular cameras.

dinarkino, 2020-06-23:
I think that @joannatarko meant that we get not full trajectory but positions of frames with some interval between them. For example, I wanted to check the quality of position estimation for Euroc Machine Hall 05 sequence, and I got 2273 poses. But there are 22212 poses in the ground truth. I was also thinking that something wrong, but then I realised that ground truth has just higher frequency and there are actually just 2273 images. Also, Evo easily synchronises it with gt:

```
evo_ape tum data.tum openvslam.txt -va --plot

Loaded 22212 stamps and poses from: data.tum
Loaded 2273 stamps and poses from: openvslam.txt
Synchronizing trajectories...
Found 2221 of max. 2273 possible matching timestamps between...
	data.tum
and:	openvslam.txt
```

squaresoft2015, 2020-08-13:
@dinarkino I tried your parameters, but the plotted curves still seems not correct:
[https://ftp.bmp.ovh/imgs/2020/08/0a5389af0aed9861.png](url)
even I tried the stereo mode, there is also obvious drift in plotted graph:
[https://ftp.bmp.ovh/imgs/2020/08/3e7607173ec3222e.png](url)