xdspacelab/openvslam #117: Usage as Library

vadixidav, 2019-09-17:
Currently, this project has a bunch of examples that are really cool, but the main OpenVSLAM library cannot be easily used because of the lack of installing, inability to use find_package, etc. It should be possible to utilize this as a library in a downstream project the same way that OpenVSLAM utilizes things like g2o, OpenCV, etc.

klintan, 2019-12-06:
Check out this PR, seems to be an effort to make it easier to use as a library https://github.com/xdspacelab/openvslam/pull/172

vadixidav, 2019-12-06:
That seems like it should help. Looking forward to that merging 👍.

shinsumicco, 2019-12-09:
Hi, @vadixidav @klintan.
I've just merged the PR from @schweitzer about improvement of CMake scripts, which enables us to install OpenVSLAM as a library.
Could you try `master` or `develop` branch?

I've confirmed the proper behavior with the following CMakeLists.txt : 

```cmake
cmake_minimum_required(VERSION 3.1)
project(openvslam LANGUAGES CXX C)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

find_package(openvslam REQUIRED)
add_executable(main main.cc)
target_link_libraries(main PRIVATE openvslam::openvslam)
```

Thanks.

klintan, 2019-12-09:
Awesome thanks! yes I saw that, I'll rebase my ROS2 PR, hopefully that'll solve some of the issues I had.