xdspacelab/openvslam #119: Trying to build ROS nodes

griz1112, 2019-09-21:
Getting this error trying to build the ROS nodes. openvslam/ros/src/openvslam/src/run_localization.cc:71:14: error: ‘class openvslam::system’ has no member named ‘track_for_monocular’

ethanguo0327, 2019-09-23:
facing the same problem ...I have no idea,some body give any help?

ricardoGrando, 2019-09-23:
Change 

SLAM.**track_for_monocular**(cv_bridge::toCvShare(msg, "bgr8")->image, timestamp, mask); 
to
SLAM.**feed_monocular_frame**(cv_bridge::toCvShare(msg, "bgr8")->image, timestamp, mask);

It will compile!

ethanguo0327, 2019-09-24:
it works! thank u so much! @ricardoGrando

shinsumicco, 2019-09-24:
Thank you for posting, @griz1112, @ethanguo0327, and @ricardoGrando.
I forgot to change the method names in the ROS module according to refactoring of  `openvslam::system` class.
I'll fix it soon.

PR #120 (2019-09-24): Modify the method name in ROS module
fix #119
commit 61fcf08f (2019-09-24): Modify the method name in ROS module

fix #119
closed 2019-09-24 by shinsumicco