xdspacelab/openvslam #121: How to Generate "orb_vocab.dbow2" Format Files ?

ZQxiaoshi, 2019-09-25:
I'm trying to train my  vocabulary，I have generated the Riyml format file，but how to generate. dbow2 format file?
Is there open source code available?

saweer, 2019-09-26:
@ZQxiaoshi
Try to use `DBoW2::TemplatedVocabulary::create` method to create an ORB vocabulary.
I did it with the KITTI sequences.
https://github.com/shinsumicco/DBoW2/blob/master/include/DBoW2/TemplatedVocabulary.h#L84

ZQxiaoshi, 2019-09-26:
@saweer 
Thank you very much for your reply.I want to convert Myvoc.txt into an "orb_vocab.dbow2" file.
 I've run the specified code and generated the executable “convert_orb_vocab”，When I enter the following command at the terminal:

./convert_orb_vocab  /Myvoc.txt   /OUT.bin  /home/zq/catkin_ws/src/DBoW2/build/   /home/zq/catkin_ws/src/DBoW2/build/

Terminal output:

Usage:
./convert_orb_vocab INPUT_TYPE:(txt|bin) OUTPUT_TYPE:(txt|bin) INPUT_FILEPATH OUTPUT_FILEPATH

Is this a question of command format? What is the correct command format?
Or something else went wrong？

saweer, 2019-09-26:
@ZQxiaoshi 
Try this:
```
./convert_orb_vocab txt bin /path/to/Myvoc.txt /path/to/OUT.bin
```

You may also create a vocabulary in binary format with `DBoW2::TemplatedVocabulary::saveToBinaryFile` method.
https://github.com/shinsumicco/DBoW2/blob/master/include/DBoW2/TemplatedVocabulary.h#L248

ZQxiaoshi, 2019-09-27:
@saweer 
Thank you very! It works correctly!

shinsumicco, 2019-11-05:
Thanks, @saweer.

closed 2019-11-05 by shinsumicco