xdspacelab/openvslam #146: Support for Odometry input

tim-fan, 2019-10-15:
Hi there, this looks like a great package - I'm excited to try it out!

I'm considering whether I could use this on a robot with a single monocular camera. In this case I imagine it would be useful to include odometry from wheel encoders, to provide scale information to the map. 

I haven't seen any mention in the documentation about odom input - is this something that is being considered for a future release? How might one go about implementing this?

MikiyaShibuya, 2019-10-18:
Thank you for your interested in our job!
The IMU odometry integration is on the way, but, unfortunately, we don't have any plan for such wheel odometry implementation.

One customization idea is adding the wheel movement distance term to the optimization.
To realize it, impose a constraint that the translation length between neighboring two keyframe equals the robot's travel distance(this modification might be something hard).

tim-fan, 2019-10-22:
Thanks, I appreciate the info. I'll keep this in mind for when I get the spare time to test this out on my robot, and let you know if I have any updates.

closed 2019-10-22 by tim-fan