xdspacelab/openvslam #147: What is the recommended way to create a mesh out of the point cloud created by openvslam?

GermanCreator, 2019-10-17:
I converted the point cloud to txt using the very useful script in https://github.com/xdspacelab/openvslam/issues/87
Having the point cloud in meshlab, I unsucessfully experimented with Poisson reconstruction to create a mesh out of it.

What is the recommended way to create a mesh out of the point cloud created by openvslam and converted with https://github.com/xdspacelab/openvslam/issues/87?

GermanCreator, 2019-10-18:
I am not happy with the result. It is either a collection of bulbs intertwined or a set of patches floating in the air. Nothing resembles the scene shown.

dirtshell, 2020-01-07:
Disclaimer: haven't actually been able to install and run the software yet, but I am researching this issue because it is of interest to me.

When creating a mesh you often times want a dense point cloud. I believe the pointcloud generated by OpenVSLAM is simply a pointcloud of the ORB features that are used for frame matching. Thus, you end up with a sparse point cloud that isn't ideal for generating meshes. 

The process of getting dense point clouds from monocular vision systems is an open research topic, but it is very doable. It is especially doable if you take the trajectory output from OpenVSLAM and use it to postprocess your video. But I don't know of any OSS that does this for you.

Zumbalamambo, 2020-05-29:
@GermanCreator have you managed to solve it?