xdspacelab/openvslam #148: Dead lock in graph_node::recover_spanning_connections

riematrix, 2019-10-18:
Hi,
  We encountered a dead lock problem after a short period of time. We just move a camera around a certain location with a radius less than 1 meter. After several round the program was blocked. The tracking and mapping thread hanged at "update_local_keyframes" and "change_spanning_parent", and below is the detailed call stack :
[image]
[image]

On debuging the code we found that the code at graph_node.cc around line 304 may be the cause. One keyframe spanning_child and spanning_parent_ is the same pointer, which is not reasonable. 

Thanks.

Dylanooo, 2019-12-07:
I have the same problem.

shinsumicco, 2019-12-07:
@riematrix and @Dylanooo 
Thank you for reporting. We'll inspect it.
Could you tell us the detail instruction to reproduce the issue if possible?

Masahiro1002, 2020-07-06:
Is this problem solved?
I have the same problem.