xdspacelab/openvslam #149: Colored point cloud with export

GermanCreator, 2019-10-18:
Have a colored point cloud that is also exportable to ply.

vwvw, 2019-10-24:
I started working on something similar a few days ago if you want to have a look. I still encounter a segfault that I didn't had time to investigate.
[colors](https://github.com/vwvw/openvslam/tree/colors)

reunanen, 2020-01-04:
> I still encounter a segfault that I didn't had time to investigate.

@vwvw tried to fix the segfault; created a [PR](https://github.com/vwvw/openvslam/pull/1/files) for you.

There were two separate issues I think:
- Can't extract color from grayscale images.
- `cv::Mat`'s `at` function's arguments are (_row_, _column_) (and not (_column_, _row_)) – but even easier is to use the point object (`pt`) as it is!

Seems to work ok now:
[image]

vwvw, 2020-01-07:
Thanks for the help! It looks good to me but didn't took much time to test it. Currently a bit busy

reunanen, 2020-01-30:
@vwvw Would be nice to create a PR of this, I think. If you are busy, I can do it as well?

vwvw, 2020-01-30:
Yeah, no much time in sight! Happy if you can do it!