xdspacelab/openvslam #150: g2o choice

cdcseacave, 2019-10-18:
Thank you for publishing this great library. I was just wandering what is your opinion on `g2o` library, and if it would be beneficial transitioning to something like https://github.com/dongjing3309/minisam or https://gtsam.org ?

shinsumicco, 2019-11-05:
@cdcseacave 
Thank you for your advice.
In fact, we first tried to use CeresSolver to utilize its autodiff. But some problems about scale drift could not be solved, so `g2o` is used at the moment.
Do you have any information about `minisam`'s and `gtsam`'s performance?

P.S. I sometimes use your OpenMVS.

cdcseacave, 2019-11-05:
No, I don't, in fact I was hoping you do and this question was more for me to learn about various solvers. I have experience only with Ceres. However I heard some good things about gtsam, and minisam looks promising because it is light.