xdspacelab/openvslam #151: FBoW issue

Shashika007, 2019-10-21:
Hi,

It is a great work! I wanted to run FBoW instead of DBoW2. I used https://github.com/rmsalinas/fbow version of fbow. When i try to build openvslam by -DBOW_FRAMEWORK=FBoW , it leads to following error. (Note - openvslam detects FBoW in the system)

error: ‘BoWVector’ in namespace ‘fbow’ does not name a type
     fbow::BoWVector bow_vec_;
error: ‘BoWFeatVector’ in namespace ‘fbow’ does not name a type
     fbow::BoWFeatVector bow_feat_vec_;

Do you use any custom version of FBoW ? or did you do any modification on that repo. 
Can you explain how to build and run with FBoW ? well appreciate. 

Best regards, 
Shashika

Shashika007, 2019-10-21:
fixed it by building this repo https://github.com/shinsumicco/FBoW.

closed 2019-10-21 by Shashika007

Gilaine, 2020-08-06:
Excuse me, when I try to build Openvslam using FBoW, it gives me these errors 
../lib/libopenvslam.so: undefined reference to `fbow::BoWVector::score(fbow::BoWVector const&, fbow::BoWVector const&)'
../lib/libopenvslam.so: undefined reference to `fbow::Vocabulary::~Vocabulary()'
../lib/libopenvslam.so: undefined reference to `fbow::Vocabulary::transform(cv::Mat const&, int, fbow::BoWVector&, fbow::BoWFeatVector&)'
../lib/libopenvslam.so: undefined reference to `fbow::Vocabulary::readFromFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
test/CMakeFiles/test_openvslam_solve_essential_solver.dir/build.make:125: recipe for target 'test/test_openvslam_solve_essential_solver' failed
make[2]: *** [test/test_openvslam_solve_essential_solver] Error 1
CMakeFiles/Makefile2:947: recipe for target 'test/CMakeFiles/test_openvslam_solve_essential_solver.dir/all' failed
make[1]: *** [test/CMakeFiles/test_openvslam_solve_essential_solver.dir/all] Error 2

Do you have any idea what would be the error?